espressif/esp-at #335: AT command learn connected user number.

hiflord, 2020-01-31:
Hello. I am doing a project with esp32 devkitc v4. I send AT commands to setup webserver and when one user connects my web server, his user number is coming to me like 1,CONNECT. After that I am using that user number for closing server. The sample picture is in below.
[image]

Sometimes two connect number is coming sometimes one connect number is coming. I didnt understand how is work the system. The sample picture two is in below.
[image]

and thinking algorithmically seems difficult to me. That is the question Is there any AT command to learn which user connected my web server? Thank you help me btw.

xcguang, 2020-02-03:
Can you capture wifi packet? It seems the new connection is built.

hiflord, 2020-02-03:
Do you mean 192.168.1.192? How can I close server with this knowladge? I wrote a algorithim. It seems working fine. I have just wondered is there any AT command that I learn which connection send request?

xcguang, 2020-02-03:
Do you send `AT+CIPSTART`?  Or you can send `AT+SYSMSG`, to print the ip and port infos if there is a connection is built.

hiflord, 2020-02-03:
Thank you it is done. But at this time my html code seems fine in android phones and windows computers. But in ios phones it seems like source code, like pure html code. What should I do for ios users?

xcguang, 2020-02-07:
I'm not sure if there is different between android and iOS, I think capturing the wifi packet is  a better way to confirm data integrity

hiflord, 2020-02-07:
okey, I will research later this problem. I can capture connected user. For now thank you so much and we can close.

closed 2020-02-07 by hiflord