espressif/esp-at #339: MQTT ESP32 setup

hiflord, 2020-02-06:
Hello. I have graduate project about IoT using with esp32. I need to setup mqtt broker with AT commands. I dont know which AT commands I should use. so Is there any document about it? I couldnt find can you help me please?

xcguang, 2020-02-07:
There are 3 blogs in Chinese, they may be helpful for you.
- AT+MQTT overview: https://blog.csdn.net/espressif/article/details/101713780
- Azure IoT: https://blog.csdn.net/espressif/article/details/103736743
- AWS IoT: https://blog.csdn.net/espressif/article/details/101714951

hiflord, 2020-02-07:
AT+MQTTUSERCFG=<LinkID>,<scheme>,<"client_id">,<"username">,<"password">,<cert_key_ID>,<CA_ID>,<"path">
I couldnt work it.
[image]
AT+MQTTUSERCFG=0,1,"","mrijxgpu","zPx3ho...",0,0,""\r\n 
Can you say what is my client_id?
[image]
I see always ERROR return. Please help me

hiflord, 2020-02-07:
Whole mqtt command gives me error. 
[image]

xcguang, 2020-02-07:
Please compile firmware using the latest master branch by yourself. 
@ustccw

hiflord, 2020-02-07:
Thank you for help @xcguang But I have latest firmware
[image]

xcguang, 2020-02-07:
The latest release firmware does not support MQTT command. Currently you have to compile it by yourself. It will be enabled by default in the next release version.

hiflord, 2020-02-07:
When will you publish new release? And is there any document to compile it by myself? Then you can close. Thank you btw

xcguang, 2020-02-07:
[image]

If you use virtual box, you can try ubuntu1804.ova, which we have setup compile environment

Or you can try the bin [artifacts-5.zip](https://github.com/espressif/esp-at/files/4170736/artifacts-5.zip)
I will close this issue, if you have any other question, please create new issue

closed 2020-02-07 by xcguang