espressif/esp-at #341: CMD0 response error, expect 0x1, response 3

YutingYou, 2020-02-09:
I've ported at_sdspi_host driver to my MCU, but sending the CMD0 by using `at_spi_cmd_go_idle_state()` doesn't return the correct response.

```
E sdspi_transaction: CMD0 response error, expect 0x1, response 3
E sdspi_transaction: Please restart slave and test again,error code:264
E sdspi_transaction: CMD0 response error, expect 0x1, response 3
E sdspi_transaction: Please restart slave and test again,error code:264
E sdspi_transaction: CMD0 response error, expect 0x1, response 3
```
[image]

Any thoughts? Thanks!

YutingYou, 2020-02-09:
As per the cmd data structure:
```
typedef struct {
    // These fields form the command sent from host to the card (6 bytes)
    uint8_t cmd_index : 6;
    uint8_t transmission_bit : 1;
    uint8_t start_bit : 1;
    uint8_t arguments[4];
    uint8_t stop_bit : 1;
    uint8_t crc7 : 7;
    /// Ncr is the dead time between command and response; should be 0xff
    uint8_t ncr;
    /// Response data, should be set by host to 0xff for read operations
    uint8_t r1;
    /// Up to 16 bytes of response. Luckily, this is aligned on 4 byte boundary.
    uint32_t response[4];
    /// response timeout, in milliseconds
    //int timeout_ms;
} sdspi_hw_cmd_t;
```
👉`Response data, should be set by host to 0xff for read operations`

The 8th bytes of MISO should output 255, in my case, I received 254. The response code should output 1 from ESP32 slave, in my case, I received 3. The interesting thing is that if we move the waveform of MISO only one bit backward, could it be the expected result?

ESP-Coco, 2020-02-10:
Hi, have you ever tried two esp32 to communicate(One ESP32 as SDIO HOST)? CMD0 response error usually hardware connection or SDIO SPI init problems.

YutingYou, 2020-02-11:
@ESP-Coco Two ESP32 work fine. But the output waveform is quite different from my MCU<->EPS32.

[image]

Jacques-zhao, 2020-02-14:
It is too strange, The communication of SDIO is controlled by the protocol and has nothing to do with the hardware platform. What MCU are you using? I am doing the migration work on the STM32 platform. Maybe I can give you some useful advice when I'm done.

YutingYou, 2020-02-16:
Thank you @Jacques-zhao, I'm using our own platform which is based on STM32 and nRF52840.

YutingYou, 2020-02-16:
@Jacques-zhao Since I've finished porting the driver, maybe we can have some more discussion about this issue. If it is convenient for you, may I have your WeChat ID, you can send it to my Email 👉: yuting0501@gmail.com. Thanks!

ESP-Coco, 2020-02-17:
I have already passed the initialization process with the STM32 development board, so I still suspect there is a problem with your hardware. The following is the log I printed with STM32.
[image]
You can send an email to our business( sales@espressif.com ) to apply for a SDIO development board. which has already debugged the hardware. You only need to use Dupont line to connect to your MCU.

YutingYou, 2020-02-18:
Thank you @ESP-Coco . Apart from the hardware issue, I also have a concern about the driver.

As per the comment inside `sdspi_host.c` driver, it should clock out 80  cycles (10 bytes) .
```
/// Clock out 80 cycles (10 bytes) before GO_IDLE command
void go_idle_clockout()
```
I did the same thing on my MCU(on the left side in the picture):
[image]

But when I use ESP32 as ESP-AT master, it clocks out only two bytes?
[image]

YutingYou, 2020-02-26:
It was the hardware issue, pretty much the EMI issue, changing the board helped.

closed 2020-02-26 by YutingYou