espressif/esp-at #343: TLS MQTT Server Setup

hiflord, 2020-02-10:
Hello. I am trying to setup MQTT server via TLS. I can create TCP MQTT server but I couldnt create TLS MQTT server. I used this at commands for setup TCP mqtt server. It is working fine

AT+MQTTUSERCFG=0,1,"mrijxgpu","mrijxgpu","zPx3hoYuBRl5",0,0,""
AT+MQTTCONN=0,"hairdresser.cloudmqtt.com",16743,0
AT+MQTTSUB=0,"topic1",1
AT+MQTTPUB=0,"topic1","test1",1,0
AT+MQTTCLEAN=0

I will try to use thesee at commands for TLS MQTT server,
AT+CIPSNTPCFG=1,8,"ntp1.aliyun.com"
AT+CIPSNTPTIME?
AT+MQTTUSERCFG=0,3,"mrijxgpu","mrijxgpu","zPx3hoYuBRl5",0,0,""
AT+MQTTCONNCFG=0,0,0,"lwtt","lwtm",0,0
AT+MQTTCONN=0,"hairdresser.cloudmqtt.com",1883,0
AT+MQTTSUB=0,"topic",1
AT+MQTTPUB=0,"topic","test",1,0
AT+MQTTCLEAN=0
But the second at commands part When I send AT+MQTTCONN=0,"hairdresser.cloudmqtt.com",1883,0
It gives me error. What is wrong? I couldnt fix it. Please help!

ustccw, 2020-02-11:
@hiflord it looks ssl handshake failed.
Had you instead of CA certificates and re-compile?
here is a demo will show you how to connect to AWS by SSL, but it was written by Chinese.
[ESP-AT 应用: AT+MQTT 对接亚马逊云](https://blog.csdn.net/espressif/article/details/101714951)

One more, you could collect the AT log for checking whether ssl handshake failed.

hiflord, 2020-02-13:
I am not good about network knowladge but here is my log
[image]
If I try to setup normal TCP server it is successfull. Here is my succesfull log
[image]
I tried AWS and it is failled too. What am I doing wrong?

hiflord, 2020-02-13:
Should I send any certificate for TLS?
[image]
[image]

hiflord, 2020-02-13:
I need document about load certificates.

ustccw, 2020-02-13:
@hiflord For easily understand, you could connect to your MQTT broker on PC firstly, via mosquitto or mqttfx.  
then try with ESP32, maybe it could help you to understand.

hiflord, 2020-02-14:
Thank you for advice. it is logical.

closed 2020-02-14 by hiflord