espressif/esp-at #344: ESP-32 AT+CWJEAP

nathandickey, 2020-02-12:
When trying to connect to the eduroam (WPA2 Enterprise) network at my school with the AT+CWJEAP command I always the "+CWJEAP:Timeout" error. Anyone know what causes this and what I need to do to be able to connect?

format: AT+CWJEAP="eduroam",1,"username@school.edu","username","password",2

xcguang, 2020-02-13:
@ustccw Please give some advice

ustccw, 2020-02-24:
@nathandickey have you replaced the CA certificate, and re-compile AT?

ustccw, 2020-02-24:
@nathandickey it looks like a certificate issue.

nathandickey, 2020-02-25:
@ustccw do I need to generate a new cert or have one provided by my school's IT? Forgot to mention earlier that my school uses PEAP/MSCHAPv2 for the eduroam network.

ustccw, 2020-02-25:
@nathandickey you could have a try with `AT+CWJEAP="eduroam",1,"username@school.edu","username","password",0`.

you could query a certificate from your school IT department if school network enables a certificate authorization.

closed 2021-04-14 by FayeY