espressif/esp-at #347: NO CA FOUND

hiflord, 2020-02-14:
I am getting this error:
[image]
I added bin files to true adress which in below.
[image]
What is my fault? Help please

hiflord, 2020-02-14:
I used these ofsets
client_cert	0x40	5	0x2a000	8K
client_key	0x40	6	0x2c000	8K
client_ca	        0x40	7	0x2e000	8K
and my at command set is
AT+CWMODE=1
AT+CIPMUX=0
AT+CWJAP="Test","helloworld"
AT+CIPSNTPCFG=1,1,"us.pool.ntp.org"
AT+CIPSNTPTIME?
AT+CIPSSLCCONF=3,0,0
and when I send 
AT+CIPSTART="SSL","a2u7s0al75g2nd-ats.iot.us-east-2.amazonaws.com",8443
It returns NO CA FOUND

xcguang, 2020-02-14:
Please do not change the partition name. And your amazon_ca looks 14KB, but the size of partition client_ca is 8K

hiflord, 2020-02-14:
What should I do for that size? What do you mean dont change the partition name? 
[image]

hiflord, 2020-02-14:
[download.txt](https://github.com/espressif/esp-at/files/4204271/download.txt)
I am using these adresses

hiflord, 2020-02-14:
I increased ofset but still same error and I changed my bin file's name.
[image]
still it is not working.
[image]
Please help

hiflord, 2020-02-17:
@xcguang @ustccw ?

hiflord, 2020-02-18:
[image]
UART1 and UART0 log

xcguang, 2020-02-18:
Please provide your at_customize.csv

hiflord, 2020-02-18:
I dont use your esp-idf library. I am connecting esp32 with stm and there is uart wire between esp32 and stm. What is relevant at_customize.csv ? Do you mean at_customize.bin?

hiflord, 2020-02-18:
I fixed it. CA certificate should download with click save as. After that move my certificate file had 1k size. .docx is making file's size too high. Thank you.

closed 2020-02-18 by hiflord