espressif/esp-at #349: CAN'T CONNECT AWS

hiflord, 2020-02-18:
Hello. I'm Ali. I couldnt connect AWS with at commands via UART1. These are my AT command order:
[image]
What is my fault here? Also how can I learn what is my client ID and name and password? please help.
I am sure to I added certificates correctly.

hiflord, 2020-02-20:
@xcguang @ustccw

ustccw, 2020-02-24:
@hiflord Hi, could you please test connection via MQTT tool, mqttfx or mosquitto or others firstly.

the log looks like a connection timeout.

hiflord, 2020-02-26:
I will try but can I ask one question? 
[image]
These are my certificates folder. client_key.bin is private.pem.key's bin extension version 
[image]
I just wanna know that I should change it with client_key.bin or mqtt_key.bin?

ustccw, 2020-02-26:
@hiflord here is a guide about how to connect to AWS by AT: https://blog.csdn.net/espressif/article/details/101714951, but it was wrotten by Chinese.
you could take a look.

xcguang, 2020-03-12:
Is the issue solved? May I close it?

hiflord, 2020-03-12:
can you wait a little more. I had some works to do.

xcguang, 2020-03-12:
If this issue is solved, I suggest you close it, and if you have any other question, you can create a new one.

closed 2020-03-17 by hiflord