espressif/esp-at #350: at_factory_bin_combine

LiThirteen, 2020-02-23:
使用make factory_bin 失败，没有出现合成后的出厂文件，先使用了make print_flash_cmd | tail -n 1 > build/download.config。 py文件中问题如下：
modlule == modlule_name  明显不成立，请问这是为什么呢
[image]

xcguang, 2020-02-26:
你是修改了 makefile 吗？怎么修改的？

LiThirteen, 2020-02-26:
@xcguang 
PROJECT_NAME := esp-at
export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP8266
export ESP_AT_MODULE_NAME ?= esp8266_default

您好，我想知道，将项目上2Mflash设置成4M需要修改哪些配置，ESP-12F，修改成4M后似乎无法获取efuse_mac，获取时错误程序无法往下执行，即使执行成功，在OTA后无法成功启动ota1分区，修改分区如下
otadata,data,ota,0x9000,8K
phy_init, data, phy,0xf000,0x1000
nvs,data,nvs,0x10000,0xE0000
at_customize,0x40,0,0xF0000,0x20000
ota_0,app,ota_0,0x110000,0x178000
ota_1,app,ota_1,0x288000,0x178000

closed 2020-02-26 by LiThirteen

xcguang, 2020-02-26:
为什么关闭了？

LiThirteen, 2020-02-26:
@xcguang   在我改动之后，factory_parameter.log这个文件似乎就开始出现了问题，文件末尾出现了.bin.bin这样不可靠的信息，所以我还原了设置，使用了WROOM-02，但明显没有找到WROOM-02，而去使用了module_config/module_esp8266_default，这样的情况下却成功了。
Makefile部分信息：
PROJECT_NAME := esp-at
export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP8266
export ESP_AT_MODULE_NAME ?= WROOM-02

控制台命令：
$ make factory_bin
Makefile:29: There is no module_config/module_wroom-02,Using module_config/module_esp8266_default

xcguang, 2020-02-26:
> export ESP_AT_MODULE_NAME ?= esp8266_default
确实是有问题的，这个后面会修复