espressif/esp-at #351: New Option ESP8266 _1M fails to run

hallard, 2020-02-24:
Just tried the new target as stated in #272 by @jandrassy parameters ESP8266_1MB

Then `make menuconfig`
in 
`AT Customized Partitions -> disable AT customized partition table`
in
`Partition Table -> Partition Table (Single factory app, no OTA)`
an then 
`Serial flasher config -> Flash size (1MB)`
`Serial flasher config -> DOUT`

Ok then 
make flash all went fine but the device failed to boot with the following error

````
(44) boot: ESP-IDF v3.3-rc1-4-g0a187af8 2nd stage bootloader
I (44) boot: compile time 00:43:28
I (44) boot: SPI Speed      : 26.7MHz
I (48) boot: SPI Mode       : DOUT
I (52) boot: SPI Flash Size : 1MB
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 000f0000
I (89) boot: End of partition table
I (94) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x94df0 (609776) map
I (432) esp_image: segment 1: paddr=0x000a4e08 vaddr=0x402a4e00 size=0x14c4c ( 85068) map
I (479) esp_image: segment 2: paddr=0x000b9a5c vaddr=0x3ffe8000 size=0x006b8 (  1720) load
I (481) esp_image: segment 3: paddr=0x000ba11c vaddr=0x40100000 size=0x00a50 (  2640) load
I (487) esp_image: segment 4: paddr=0x000bab74 vaddr=0x40100a50 size=0x058b8 ( 22712) load
I (507) boot: Loaded app from partition at offset 0x10000
phy_version: 1155.0, 6cb3053, Nov 11 2019, 17:31:08, RTOS new
partation fail
factory_parameter partition missed
partation fail
````

What did I missed?

Output of make looks ok
````
To flash all build output, run 'make flash' or:
python /Users/charles/esp/esp-at/esp-idf/components/esptool_py/esptool/esptool.py --chip esp8266 --port /dev/tty.usbserial-14410 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 26m --flash_size 1MB 0x0000 /Users/charles/esp/esp-at/build/bootloader/bootloader.bin 0x10000 /Users/charles/esp/esp-at/build/esp-at.bin 0x8000 /Users/charles/esp/esp-at/build/partitions_singleapp.bin
````

Interesting thing is that make factory bin fails also
````
charles@mac:esp-at$ make print_flash_cmd | tail -n 1 > build/download.config
charles@mac:esp-at$ make factory_bin
Toolchain path: /Users/charles/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-92-g8facf4c
Expected to see version: crosstool-ng-1.22.0-100-ge567ec7
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
Python requirements from /Users/charles/esp/esp-at/esp-idf/requirements.txt are satisfied.
0x0,bootloader/bootloader.bin
0x10000,esp-at.bin
0x8000,partitions_singleapp.bin
Traceback (most recent call last):
  File "/Users/charles/esp/esp-at/tools/esp_at_factory_bin_combine.py", line 99, in <module>
    main()
  File "/Users/charles/esp/esp-at/tools/esp_at_factory_bin_combine.py", line 96, in main
    args.flash_speed.upper(), args.bin_directory, args.parameter_file, args.download_config)
  File "/Users/charles/esp/esp-at/tools/esp_at_factory_bin_combine.py", line 65, in esp32_at_combine_bin
    addr = bin_list[default_name]
KeyError: 'factory_param.bin'
make: *** [factory_bin] Error 1
````

xcguang, 2020-02-24:
> AT Customized Partitions -> disable AT customized partition table

Why disable it? 

Please delete sdkconfig, then modify top Makefile as following and compile it again.

```
export ESP_AT_PROJECT_PLATFORM = PLATFORM_ESP8266
export ESP_AT_MODULE_NAME = ESP8266_1MB
```

Notes: DO NOT do any change in `make menuconfig`,

hallard, 2020-02-24:
That's what Ive done, restarted all from scratch, now looks like partition are fine

````
charles@mac-mini:esp-at$ make print_flash_cmd | tail -n 1 > build/download.config
charles@mac-mini:esp-at$ make factory_bin
Toolchain path: /Users/charles/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-92-g8facf4c
Expected to see version: crosstool-ng-1.22.0-100-ge567ec7
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
Python requirements from /Users/charles/esp/esp-at/esp-idf/requirements.txt are satisfied.
['0x9000 ota_data_initial.bin', '0x0000 bootloader/bootloader.bin', '0x18000 at_customize.bin', '0x19000 customized_partitions/factory_param.bin', '0x1A000 customized_partitions/client_cert.bin', '0x1B000 customized_partitions/client_key.bin', '0x1C000 customized_partitions/client_ca.bin', '0x1D000 customized_partitions/mqtt_cert.bin', '0x1E000 customized_partitions/mqtt_key.bin', '0x1F000 customized_partitions/mqtt_ca.bin', '0x20000 esp-at.bin', '0x8000 partitions_at.bin']
0x9000,ota_data_initial.bin
0x0,bootloader/bootloader.bin
0x18000,at_customize.bin
0x19000,customized_partitions/factory_param.bin
0x1a000,customized_partitions/client_cert.bin
0x1b000,customized_partitions/client_key.bin
0x1c000,customized_partitions/client_ca.bin
0x1d000,customized_partitions/mqtt_cert.bin
0x1e000,customized_partitions/mqtt_key.bin
0x1f000,customized_partitions/mqtt_ca.bin
0x20000,esp-at.bin
0x8000,partitions_at.bin
Create /Users/charles/esp/esp-at/build/factory/factory_ESP8266_1MB.bin for ESP8266_1MB finished
````

I don't have the error message for partition failed on debug but I can't see or send any AT commands, looks like nothing is happening.

Would someone publish a compiled version of 1MB factory bin working so I can flash and check? This is driving me mad from some days now

Thanks

xcguang, 2020-02-24:
I guess the the uart pins you send command are incorrect.
Please refer to https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md and https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Factory_Parameter_Bin.md

hallard, 2020-02-24:
As you suggested, I did not changed anything on menucnofig, UART pins should be RX0 TX0 default Serial of ESP not? The same used to flash the device

xcguang, 2020-02-24:
No, as described in https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md, 
[image]

hallard, 2020-02-24:
@xcguang you saved my life, it's working fine now, thank you very much, 

To be honest It's very confusing, UART0 on ESP8266 and WROOM-02 or NodeMCU or WeMos D1 Mini is always mentioned to be U0TXD/U0RXD or RXD0/TXD0 so if I follow this notation I expected UART0 to be on pin 1 and 3. Worth mentioning that on the readme.

[image]

[image]

[image]

xcguang, 2020-02-24:
you can change it as https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md

closed 2020-02-24 by xcguang

Aukstkalnis, 2021-09-28:
> I guess the the uart pins you send command are incorrect.
> Please refer to https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md and https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Factory_Parameter_Bin.md

These links do not work anymore.

xcguang, 2021-09-28:
The new links: https://docs.espressif.com/projects/esp-at/en/latest/index.html