espressif/esp-at #352: mDNS set by AT+MDNS stop responding after some time

kurta999, 2020-02-25:
I normally set mDNS with AT+MDNS, everything works but after a few minutes I need to re-start the my device (to reinit mDNS) to be visible again.

I checked mDNS myself inside LwIP and it has same behaviour too. Is this problem inside lwip or why it happens? I use Bonjour Browser for client.

SUNLUXIN, 2020-03-10:
Can you describe your case in detail and provide some log and the commands you send?

kurta999, 2020-03-15:
ATE0
AT+CWAUTOCONN=0
AT+GMR
AT+CIPMUX=1
AT+CWMODE=3
"AT+CWSAP=\"%s\",\"%s\",%d,%d,%d,%d", ssid, pwd, channel, (uint8_t)encryption,
"AT+CWHOSTNAME=\"%s\"", hostname);
"AT+CIPAP=\"%s\",\"%s\",\"%s\"", ip_address, gateway, netmask);
AT+CIPSERVERMAXCONN=3
AT+CIPSERVER=1,80
"AT+CIPSNTPCFG=1,%d,\"%s\",\"%s\",\"%s\"", timezone, svr_0, svr_1, svr_2);
"AT+MDNS=1,\"%s\",\"%s\",%d", hostname, service_name, port);

But the interesting part is that I checked it myself with LwIP on nucleo board with ethernet connector and it behaves same (I just worked with lwip with STM32, I needed it for another project). mDNS stops responding after a few minutes. Might the problem inside the lwip stack? If somebody would recommend me a way how to test this issue, I would try it. I use bonjour, this is the one thing what I know to work with mDNS.

ustccw, 2020-03-27:
@kurta999 Could you please providing the packets for us.

it looks like can not reproduce, at least in my test.

kurta999, 2020-04-01:
Could you describe a little bit how I should do it?

ustccw, 2020-04-22:
@kurta999 capture the packets by wireshark or omnipeek, etc..
Looks like the following packet in my test:

[mdns.zip](https://github.com/espressif/esp-at/files/4513601/mdns.zip)

closed 2020-08-18 by xcguang