espressif/esp-at #353: AT+HTTPCLIENT Error

hallard, 2020-02-26:
I'm using the current build on ESP8285 (set ESP8266_1B), everything looks working fine except the AT+HTTPCLIENT, none of the sample above works
https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md#cmd-HTTPCLIENT
connexion looks like ok, very strange. looks like the command does not exist

```
01:43:36.046 -> AT+CIPSTA?
01:43:36.046 -> +CIPSTA:ip:"192.168.1.179"
01:43:36.046 -> +CIPSTA:gateway:"192.168.1.1"
01:43:36.046 -> +CIPSTA:netmask:"255.255.255.0"
01:43:36.046 -> 
01:43:36.046 -> OK
01:44:21.287 -> AT+HTTPCLIENT=2,0,,"httpbin.org","/get",1
01:44:21.287 -> 
01:44:21.287 -> ERROR
01:44:30.402 -> AT+HTTPCLIENT=2,0,"http://httpbin.org/get",,,0
01:44:30.402 -> 
01:44:30.402 -> ERROR
01:44:37.689 -> AT+HTTPCLIENT=2,0,"http://httpbin.org/get","httpbin.org","/get",1
01:44:37.689 -> 
01:44:37.689 -> ERROR
01:44:44.596 -> AT+HTTPCLIENT=1,0,"httpbin.org","/get",1
01:44:44.596 -> 
01:44:44.596 -> ERROR
01:44:50.369 -> AT+HTTPCLIENT=1,0,"http://httpbin.org/get",,,0
01:44:50.369 -> 
01:44:50.369 -> ERROR
01:44:56.132 -> AT+HTTPCLIENT=1,0,"http://httpbin.org/get","httpbin.org","/get",1
01:44:56.169 -> 
01:44:56.169 -> ERROR
01:44:58.778 -> AT+CIPSTA?
01:44:58.778 -> +CIPSTA:ip:"192.168.1.179"
01:44:58.778 -> +CIPSTA:gateway:"192.168.1.1"
01:44:58.778 -> +CIPSTA:netmask:"255.255.255.0"
01:44:58.778 -> 
01:44:58.778 -> OK
01:47:06.870 -> AT+PING="www.baidu.com"
01:47:07.158 -> +PING:238
01:47:07.158 -> 
01:47:07.158 -> OK
01:47:18.111 -> AT+PING="google.com"
01:47:18.182 -> +PING:34
01:47:18.182 -> 
01:47:18.182 -> OK
```

songjiaxiu-espressif, 2020-02-26:
Please enable ` AT http command support` function in the `make menuconfig`.
`make menuconfig` --> `Component config` --> ` AT` --> ` AT http command support`

[image]

hallard, 2020-02-26:
@songjiaxiu-espressif thank you 
Still don't know how I missed that one, works much better ;-)

closed 2020-02-26 by hallard