espressif/esp-at #354: 编译工程报:undefined reference to `SC_EVENT'

woshi520laoxie, 2020-02-26:
前提：之前已经搭建好环境
1、目前重新创建一个新的work目录 /work
1.1、github下载一个最新的文件  esp-at-master
         路径为:/work/esp-at-master
1.2、复制esp-idf文件夹文件到   /work下面
         /work/esp-idf
1.3、设置好编译器路径
1.4、执行make menuconfig
1.5、报错:
/work/esp-at-master/components/at/lib/libesp32_at_core.a(at_smartconfigCmd.c.obj):(.literal.esp_at_smartconfig_cmd_regist+0x4): undefined reference to `SC_EVENT'
collect2: error: ld returned 1 exit status
1.6、问 --- 该错误的原因是什么呢？

ustccw, 2020-02-26:
@woshi520laoxie Please compile the `esp-at` according to [README.md](https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md)

woshi520laoxie, 2020-02-26:
@ustccw 你好:我目前已经搭建好一份环境了，已经编译过一份之前的esp-at工程

woshi520laoxie, 2020-02-26:
@ustccw

woshi520laoxie, 2020-02-26:
@ustccw 你好，我之前已经编译好了一份 esp-at 工程，工具链和需要的环境都已经完成
1、目前重新建立一个文件夹，然后下载了一份新的esp_at工程，以之前的方式编译，却报错了。
2、其实最好的状态应该是和u-boot类似，定义好交叉编译工具，然后就可以直接编译了。

woshi520laoxie, 2020-02-26:
@ustccw 你好，我之前已经编译好了一份 esp-at 工程，工具链和需要的环境都已经完成
1、目前重新建立一个文件夹，然后下载了一份新的esp_at工程，以之前的方式编译，却报错了。
2、其实最好的状态应该是和u-boot类似，定义好交叉编译工具，然后就可以直接编译了。
3、刚才我执行了 make defconfig之后再次执行 make menuconfig
4、可以从我的提示信息看得出来 异常的原因吗？
------   截取最后部分信息
Toolchain path: /home/sambaShare/esp/xtensa-esp32-elf/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
GENCONFIG

*** Error during update of the configuration.
make: *** 没有规则可制作目标“/home/work/esp-at-master/build/include/config/auto.conf”，由“/home/work/esp-at-master/build/bootloader/bootloader.bin” 需求。 停止。

ustccw, 2020-02-27:
@woshi520laoxie Please run `sudo rm -rf build sdkconfig sdkconfig.old` firstly, and then run `make menuconfig && make`.

woshi520laoxie, 2020-02-28:
@ustccw 好的，谢谢

xcguang, 2020-03-12:
问题解决了吗？

closed 2021-04-14 by FayeY