espressif/esp-at #356: 2 Routers with same SSID/Password, ESP keeps IP Setting of the other AP

hallard, 2020-02-26:
I got 2 Routers, one at home and one at work office, both have the same SSID/Password so I'm able to bring and drive devices from home/office without any issue.

both routers are on the network 192.168.1.0/24 the only difference between home and office if that office GW/DNS is 192.168.1.254 and at home it's 192.168.1.1

But with this at-firmware, something strange is happening, I mean when connect 1st time (here below at job) I got connexion and then all is fine (see below) and when I'm going home, when the device start it connect and got IP but keeps IP settings of the previous router, this mean it has wrong IP for DNS and gateway but also keep it's own IP (that can conflict with another one on the same network), not the one distributed by the router. Looks like if SSID Network is the same, it keeps First config as "static". That's why I forced DHCP in the following log

Issue from office with fresh device today.
```
16:43:52.673 -> AT
16:43:52.673 -> 
16:43:52.673 -> OK
16:45:11.488 -> AT+CWMODE=1
16:45:11.488 -> 
16:45:11.488 -> OK
16:45:20.301 -> AT+CWDHCP=1,1
16:45:20.301 -> 
16:45:20.301 -> OK
16:45:33.035 -> AT+CIPSTA?
16:45:33.035 -> +CIPSTA:ip:"0.0.0.0"
16:45:33.035 -> +CIPSTA:gateway:"0.0.0.0"
16:45:33.035 -> +CIPSTA:netmask:"0.0.0.0"
16:45:33.035 -> 
16:45:33.035 -> OK
16:53:52.868 -> AT+CWJAP="MY-WIFIAP","ThisIsMyPassword"
16:53:55.330 -> WIFI CONNECTED
16:53:58.861 -> WIFI GOT IP
16:53:58.861 -> 
16:53:58.861 -> OK
16:55:13.149 -> AT+CIPSTA?
16:55:13.149 -> +CIPSTA:ip:"192.168.1.46"
16:55:13.149 -> +CIPSTA:gateway:"192.168.1.254"
16:55:13.149 -> +CIPSTA:netmask:"255.255.255.0"
16:55:13.149 -> 
16:55:13.149 -> OK
```
Then I plugged the one configured at home last evening, hoping the correct configuration
```
17:23:39.440 -> AT+RESTORE
17:23:39.476 -> WIFI DISCONNECT
17:23:40.145 -> 
17:23:40.145 -> OK
17:23:40.212 ->  l⸮⸮p| p⸮p lp⸮lp ⸮⸮|⸮ l⸮l⸮lp ⸮⸮⸮⸮|l⸮lp ⸮⸮⸮⸮|l⸮ll
17:23:41.143 -> ready
17:24:07.925 -> AT+CWMODE=1
17:24:07.962 -> 
17:24:07.962 -> OK
17:24:13.858 -> AT+CWDHCP=1,1
17:24:13.858 -> 
17:24:13.858 -> OK
17:24:23.420 -> AT+CIPSTA?
17:24:23.420 -> +CIPSTA:ip:"0.0.0.0"
17:24:23.420 -> +CIPSTA:gateway:"0.0.0.0"
17:24:23.420 -> +CIPSTA:netmask:"0.0.0.0"
17:24:23.420 -> 
17:24:23.420 -> OK
17:26:35.609 -> AT+CWJAP="MY-WIFIAP","ThisIsMyPassword"
17:26:38.023 -> WIFI CONNECTED
17:26:40.557 -> WIFI GOT IP
17:26:40.557 -> 
17:26:40.557 -> OK
17:26:44.049 -> AT+CIPSTA?
17:26:44.049 -> +CIPSTA:ip:"192.168.1.179"
17:26:44.049 -> +CIPSTA:gateway:"192.168.1.1"
17:26:44.083 -> +CIPSTA:netmask:"255.255.255.0"
17:26:44.083 -> 
17:26:44.083 -> OK
```

As you can see the configuration is wrong ? the AP even distributed any DHCP address because I'm pretty sure the device keeps static IP settings.

Any idea of what could cause this issue?

SUNLUXIN, 2020-03-10:
Hi, 
Can you provide version information? thanks

hallard, 2020-03-10:
Sure,
It’s the latest master 

Hi,
Can you provide version information? thanks

SUNLUXIN, 2020-03-11:
Hi,
I used the latest master and did not reproduce the issue. Can you directly provide me with firmware to reproduce this issue?
I have two routers configured the same as yours, and before the device connects to the first router, I power off the second router. After the device was connected to the first router, I started to power off the first router. Power on the second router, and the device started to connect to the second router. Looking at the console of the second router, I can see that the device is assigned a DHCP address
//
[2020-03-11_10:07:06]AT+RESTORE
[2020-03-11_10:07:06]WIFI DISCONNECT
[2020-03-11_10:07:06]OK
[2020-03-11_10:07:08]ready
[2020-03-11_10:07:16]AT+GMR
[2020-03-11_10:07:16]AT version:2.1.0.0-dev(20b4946 - Mar  4 2020 10:24:39)
[2020-03-11_10:07:16]SDK version:v4.0-174-gf9cb434
[2020-03-11_10:07:16]compile time(f1f1746):Mar  4 2020 11:54:03
[2020-03-11_10:07:16]Bin version:2.0.0(WROVER-32)
[2020-03-11_10:07:16]OK
[2020-03-11_10:09:02]AT+CWMODE=1
[2020-03-11_10:09:02]OK
[2020-03-11_10:09:16]AT+CWDHCP=1,1
[2020-03-11_10:09:16]OK
[2020-03-11_10:09:39]AT+CIPSTA?
[2020-03-11_10:09:39]+CIPSTA:ip:"0.0.0.0"
[2020-03-11_10:09:39]+CIPSTA:gateway:"0.0.0.0"
[2020-03-11_10:09:39]+CIPSTA:netmask:"0.0.0.0"
[2020-03-11_10:09:39]OK
[2020-03-11_10:11:13]AT+CWJAP="sunluxin_ap","1234567890"
[2020-03-11_10:11:17]WIFI CONNECTED
[2020-03-11_10:11:18]WIFI GOT IP
[2020-03-11_10:11:18]OK
[2020-03-11_10:11:25]AT+CWJAP?
[2020-03-11_10:11:25]+CWJAP:"sunluxin_ap","6c:e8:73:45:88:ae",10,-28,0,0,0
[2020-03-11_10:11:25]OK
[2020-03-11_10:11:39]AT+CIPSTA?
[2020-03-11_10:11:39]+CIPSTA:ip:"192.168.1.101"
[2020-03-11_10:11:39]+CIPSTA:gateway:"192.168.1.1"
[2020-03-11_10:11:39]+CIPSTA:netmask:"255.255.255.0"
[2020-03-11_10:11:39]OK

[2020-03-11_10:12:21]AT+RESTORE
[2020-03-11_10:12:21]WIFI DISCONNECT
[2020-03-11_10:12:22]OK
[2020-03-11_10:12:24]ready
[2020-03-11_10:12:26]AT+CWMODE=1
[2020-03-11_10:12:26]OK
[2020-03-11_10:12:35]AT+CWDHCP=1,1
[2020-03-11_10:12:35]OK
[2020-03-11_10:12:38]AT+CWJAP="sunluxin_ap","1234567890"
[2020-03-11_10:12:42]WIFI CONNECTED
[2020-03-11_10:12:43]WIFI GOT IP
[2020-03-11_10:12:43]OK
[2020-03-11_10:12:56]AT+CIPSTA?
[2020-03-11_10:12:56]+CIPSTA:ip:"192.168.1.100"
[2020-03-11_10:12:56]+CIPSTA:gateway:"192.168.1.254"
[2020-03-11_10:12:56]+CIPSTA:netmask:"255.255.255.0"
[2020-03-11_10:12:56]OK
[2020-03-11_10:13:08]AT+CWJAP?
[2020-03-11_10:13:08]+CWJAP:"sunluxin_ap","9c:a6:15:82:db:83",8,-20,0,0,0
[2020-03-11_10:13:08]OK
//

hallard, 2020-03-12:
@SUNLUXIN 
So I trusted your tests and changed my office router, installed same LAN config as 10 years old one router and guess what, now everything is working fine. Sorry about that still don't understand what happened but for sure was time to upgrade my office router.

closed 2020-03-12 by hallard

xcguang, 2020-03-12:
I guess, it may be the DHCP info the second router responds does not include GW information, and the recorded GW information is not updated. Can you capture the DHCP packet?

hallard, 2020-03-12:
I could, but I'm so lack of time for now. Anyway, I'm pretty sure it's not a bug (at least from this repo ;-)