espressif/esp-at #357: 执行idf.py build:报异常ninja: build stopped: subcommand failed.

woshi520laoxie, 2020-02-28:
1、分支为：esp-at-master
2、能够执行make menuconfig，可以出现图形界面
3、如果执行 idf.py menuconfig 出现异常:
      ninja: build stopped: subcommand failed.
      ninja failed with exit code 1
4、执行 idf.py build 也是包 3位置的异常

这个异常有处理方式吗？我百度过一些，异常还在，如下面这个设置
数据段长度无限制：ulimit -d unlimited
最大内存大小无限制：ulimit -m unlimited
堆栈大小无限制：ulimit -s unlimited
core文件大小无限制：ulimit -c unlimited

xcguang, 2020-02-28:
你的是什么操作系统，有详细的 错误 log 吗？发给我们看一下

woshi520laoxie, 2020-02-28:
1、使用系统版本:Ubuntu 18.04.2 LTS \n \l
&nbsp; &nbsp; &nbsp;我邮件中添加了附件，附件汇总有一些信息，之前发送有时候附件看不到的情况，你看看能不能下载这个附件

woshi520laoxie, 2020-02-28:
@xcguang 
1、使用系统版本:Ubuntu 18.04.2 LTS \n \l

2、运行指令:
---------------------------------------------
cd ./esp-idf
./install.sh
. ./export.sh
xtensa-esp32-elf-gcc -v
idf.py menuconfig
---------------------------------------------

3、信息如下:
---------------------------------------------
---------------------------------------------
工程路径:/home/sambaShare/work/esp-at-master
1
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Installing Python environment and packages
15:26:02.122627 git.c:344               trace: built-in: git describe --tags
Installing Python packages from /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Requirement already satisfied: setuptools in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 4)) (44.0.0)
Requirement already satisfied: click>=5.0 in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 8)) (7.0)
Requirement already satisfied: pyserial>=3.0 in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 11)) (2.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 13)) (0.26)
Requirement already satisfied: ipaddress; python_version < "3" in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 11)) (1.0.23)
Requirement already satisfied: six>=1.4.1 in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: enum34; python_version < "3" in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 11)) (1.1.9)
Requirement already satisfied: pycparser in /home/xwz/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt (line 11)) (2.19)
All done! You can now run:

  . ./export.sh

Adding ESP-IDF tools to PATH...
15:26:02.545376 git.c:344               trace: built-in: git describe --tags
Checking if Python packages are up to date...
Python requirements from /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /home/sambaShare/work/esp-at-master/esp-idf/components/esptool_py/esptool
  /home/sambaShare/work/esp-at-master/esp-idf/components/espcoredump
  /home/sambaShare/work/esp-at-master/esp-idf/components/partition_table/
  /home/xwz/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin
  /home/xwz/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin
  /home/xwz/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin
  /home/xwz/.espressif/python_env/idf4.0_py2.7_env/bin
  /home/sambaShare/work/esp-at-master/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/xwz/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf - [... cut ...] are/work/esp-at-master/esp-idf/Kconfig --sdkconfig-rename /home/sambaShare/work/esp-at-master/esp-idf/sdkconfig.rename --config /home/sambaShare/work/esp-at-master/sdkconfig --defaults /home/sambaShare/work/esp-at-master/module_config/module_esp32_default/sdkconfig.defaults --env-file /home/sambaShare/work/esp-at-master/build/config.env --env IDF_TARGET=esp32 --output config /home/sambaShare/work/esp-at-master/sdkconfig
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

4、结束

xcguang, 2020-02-28:
看不到附件，你可以别在邮件发，直接在 github 上回复，估计是 github 和 你们的邮件系统 对附件处理方式不一致

woshi520laoxie, 2020-02-28:
我已经直接在github上面发了，你看看

woshi520laoxie, 2020-03-03:
你好啊，我已经将部分信息直接回复到了github上了，能看得出一些问题吗？

xcguang, 2020-03-03:
当前 AT 对 Cmake 支持不太好，可以参考  https://github.com/espressif/esp-at/blob/master/.gitlab-ci.yml ，先 export 一下这些变量

```
export ESP_AT_PROJECT_PLATFORM=PLATFORM_ESP32
export ESP_AT_MODULE_NAME=WROOM-32
export ESP_AT_PROJECT_PATH=$(pwd)
```

woshi520laoxie, 2020-03-06:
@xcguang 
你好啊，按照你说的方式:
ESPAT_PROJ_PATH=$(pwd)
cd ./esp-idf
./install.sh
. ./export.sh
cd $ESPAT_PROJ_PATH
xtensa-esp32-elf-gcc -v
export ESP_AT_PROJECT_PLATFORM=PLATFORM_ESP32
export ESP_AT_MODULE_NAME=WROOM-32
export ESP_AT_PROJECT_PATH=$(pwd)
idf.py build

依旧是报这个错误:
error adding symbols: no more archived files
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

xcguang, 2020-03-06:
我刚才又试了，在我这是可以的；你提供一下详细的log，上面那点log看不到详细信息；另外你看一下 `ninja --version` 版本多少？ 你还可以，直接删除 build 目录，直接 make 试试

woshi520laoxie, 2020-03-06:
@xcguang 
$ ninja --version
1.8.2

执行makemenuconfig
可以显示配置界面
[image]

如果执行make
C build/bootloader/efuse/esp32/esp_efuse_table.o
CC build/bootloader/efuse/src/esp_efuse_api.o
CC build/bootloader/efuse/src/esp_efuse_utility.o
CC build/bootloader/efuse/src/esp_efuse_fields.o
AR build/bootloader/efuse/libefuse.a
LD build/bootloader/bootloader.elf
esptool.py v2.8
building AT customized partition table module_config/module_esp32_default/at_customize.csv
building AT customized partition table succeed
generating AT customized partition bins to /home/sambaShare/work/esp-at-master/build/customized_partitions
generating ble_data.bin: python /home/sambaShare/work/esp-at-master/tools/BLEService.py -t /home/sambaShare/work/esp-at-master/build/customized_partitions/ble_data.bin /home/sambaShare/work/esp-at-master/components/customized_partitions/raw_data/ble_data/example.csv
Traceback (most recent call last):
  File "/home/sambaShare/work/esp-at-master/tools/BLEService.py", line 30, in <module>
    import xlrd
ImportError: No module named xlrd
/home/sambaShare/work/esp-at-master/components/customized_partitions/Makefile.projbuild:38: recipe for target '/home/sambaShare/work/esp-at-master/build/at_customize.bin' failed
make: *** [/home/sambaShare/work/esp-at-master/build/at_customize.bin] Error 1

import xlrd
ImportError: No module named xlrd

安装 xlrd
pip install xlrd
Collecting xlrd
  Using cached https://files.pythonhosted.org/packages/b0/16/63576a1a001752e34bf8ea62e367997530dc553b689356b9879339cf45a4/xlrd-1.2.0-py2.py3-none-any.whl
Installing collected packages: xlrd
Successfully installed xlrd-1.2.0

之后再执行make 依旧报这个错误。

我试过  直接删除 build 目录，直接 make 试试，必须编译一次 make menuconfig吧
1、我是删除 build   
2、执行 make menuconfig
3、再make  就报这个错误了

xcguang, 2020-03-06:
难不成你电脑上存在两个 python 版本？麻烦再 `pip3 install xlrd` 试一下？

woshi520laoxie, 2020-03-06:
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ pip3 install xlrd

Command 'pip3' not found, but can be installed with:

sudo apt install python3-pip

那么我需要安装这个 pip3吗？
有python3的

woshi520laoxie, 2020-03-06:
@xcguang 
wz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ python3 -V
Python 3.6.8
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ python2 -V
Python 2.7.17

xcguang, 2020-03-06:
要不安装 pip3 试一下？我用 Python 2.7.x 和 Python 3.8.0 都测试过

ustccw, 2020-03-06:
@woshi520laoxie maybe you could try the following steps:
1. remove `~/.espressif`
```
sudo rm -rf ~/.espressif
```

2. download compiler 8.2.0
https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/linux-setup.html

3. export your compiler to system environment
if success, you would get the info like:
```
chenwu@espressif:~$ xtensa-esp32-elf-gcc -v
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/opt/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-2019r2' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp-2019r2) 
```

4. install the some tools
```
sudo apt-get install gcc git wget make flex bison gperf python python-pip python-setuptools python-serial python-cryptography python-future python-pyparsing python-pyelftools libffi-dev libssl-dev
```

5. compiler AT
```
make
```

Good luck.

woshi520laoxie, 2020-03-09:
@xcguang    你好啊，目前我又测试了一下，这次报的错误换了

1、使用 pip3 install xlrd
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ pip3 install xlrd
Collecting xlrd
  Using cached https://files.pythonhosted.org/packages/b0/16/63576a1a001752e34bf8ea62e367997530dc553b689356b9879339cf45a4/xlrd-1.2.0-py2.py3-none-any.whl
Installing collected packages: xlrd

2、再次执行编译
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/xwz/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-2019r2' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp-2019r2) 
16:11:16.733834 git.c:344               trace: built-in: git rev-parse --short HEAD
fatal: 不是一个 git 仓库（或者任何父目录）：.git
Makefile:29: There is no module_config/module_wroom-32,Using module_config/module_esp32_default
16:11:16.742970 git.c:344               trace: built-in: git rev-parse HEAD
16:11:16.752186 git.c:344               trace: built-in: git describe --always --tags --dirty
16:11:16.779298 git.c:344               trace: built-in: git status --porcelain --ignore-submodules=dirty
Toolchain path: /home/xwz/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2019r2
Compiler version: 8.2.0
Python requirements from /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt are satisfied.
16:11:16.954646 git.c:344               trace: built-in: git describe --always --tags --dirty
16:11:16.969326 git.c:344               trace: built-in: git status --porcelain --ignore-submodules=dirty

App "esp-at" version: ESP-AT
LD build/esp-at.elf
/home/xwz/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/sambaShare/work/esp-at-master/components/at/lib/libesp32_at_core.a: error adding symbols: no more archived files
collect2: error: ld returned 1 exit status
/home/sambaShare/work/esp-at-master/esp-idf/make/project.mk:528: recipe for target '/home/sambaShare/work/esp-at-master/build/esp-at.elf' failed
make: *** [/home/sambaShare/work/esp-at-master/build/esp-at.elf] Error 1

woshi520laoxie, 2020-03-09:
@xcguang 
如果我先执行make menuconfig   在执行make的话 ，报的异常依旧是
ImportError: No module named xlrd
但是我已经安装了 xlrd了
1、安装  pip3 install xlrd
wz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ pip3 install xlrd
Collecting xlrd
  Using cached https://files.pythonhosted.org/packages/b0/16/63576a1a001752e34bf8ea62e367997530dc553b689356b9879339cf45a4/xlrd-1.2.0-py2.py3-none-any.whl
Installing collected packages: xlrd
Successfully installed xlrd-1.2.0
2、编译
.....
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
CC build/bootloader/efuse/esp32/esp_efuse_table.o
CC build/bootloader/efuse/src/esp_efuse_api.o
CC build/bootloader/efuse/src/esp_efuse_utility.o
CC build/bootloader/efuse/src/esp_efuse_fields.o
AR build/bootloader/efuse/libefuse.a
LD build/bootloader/bootloader.elf
esptool.py v2.8
building AT customized partition table module_config/module_esp32_default/at_customize.csv
building AT customized partition table succeed
generating AT customized partition bins to /home/sambaShare/work/esp-at-master/build/customized_partitions
generating ble_data.bin: python /home/sambaShare/work/esp-at-master/tools/BLEService.py -t /home/sambaShare/work/esp-at-master/build/customized_partitions/ble_data.bin /home/sambaShare/work/esp-at-master/components/customized_partitions/raw_data/ble_data/example.csv
Traceback (most recent call last):
  File "/home/sambaShare/work/esp-at-master/tools/BLEService.py", line 30, in <module>
    import xlrd
ImportError: No module named xlrd
/home/sambaShare/work/esp-at-master/components/customized_partitions/Makefile.projbuild:38: recipe for target '/home/sambaShare/work/esp-at-master/build/at_customize.bin' failed
make: *** [/home/sambaShare/work/esp-at-master/build/at_customize.bin] Error 1

ustccw, 2020-03-10:
@woshi520laoxie is there a directory: `~/.espressif`.

If exist, please delete it.

woshi520laoxie, 2020-03-10:
@ustccw 是的
1、我已经执行了  sudo rm -rf ~/.espressif

2、gcc 版本为:
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp-2019r2) 

3、安装一些工具
sudo apt-get install gcc git wget make flex bison gperf python python-pip python-setuptools python-serial python-cryptography python-future python-pyparsing python-pyelftools libffi-dev libssl-dev

4、再执行make
....
CC build/wpa_supplicant/src/wps/wps_dev_attr.o
CC build/wpa_supplicant/src/wps/wps.o
CC build/wpa_supplicant/src/wps/wps_common.o
CC build/wpa_supplicant/src/wps/wps_registrar.o
CC build/wpa_supplicant/src/wps/wps_attr_build.o
CC build/wpa_supplicant/src/wps/wps_enrollee.o
AR build/wpa_supplicant/libwpa_supplicant.a
CC build/xtensa/debug_helpers.o
CC build/xtensa/trax.o
AR build/xtensa/libxtensa.a
Generating esp32.project.ld
LD build/esp-at.elf
/home/xwz/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/sambaShare/work/esp-at-master/components/at/lib/libesp32_at_core.a: error adding symbols: no more archived files
collect2: error: ld returned 1 exit status
/home/sambaShare/work/esp-at-master/esp-idf/make/project.mk:528: recipe for target '/home/sambaShare/work/esp-at-master/build/esp-at.elf' failed
make: *** [/home/sambaShare/work/esp-at-master/build/esp-at.elf] Error 1

woshi520laoxie, 2020-03-10:
@ustccw 
1、删除 ~/.espressif
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ sudo rm -rf ~/.espressif
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ ls ~/
Arduino  esp_user  src  公共的  模板  视频  图片  文档  下载  音乐  桌面

2、安装工具
sudo apt-get install gcc git wget make flex bison gperf python python-pip python-setuptools python-serial python-cryptography python-future python-pyparsing python-pyelftools libffi-dev libssl-dev

安装信息如下：
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ sudo apt-get install gcc git wget make flex bison gperf python python-pip python-setuptools python-serial python-cryptography python-future python-pyparsing python-pyelftools libffi-dev libssl-dev
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
bison 已经是最新版 (2:3.0.4.dfsg-1build1)。
flex 已经是最新版 (2.6.4-6)。
libffi-dev 已经是最新版 (3.2.1-8)。
make 已经是最新版 (4.1-9.1ubuntu1)。
python 已经是最新版 (2.7.15~rc1-1)。
python-future 已经是最新版 (0.15.2-4ubuntu2)。
python-pyparsing 已经是最新版 (2.2.0+dfsg1-2)。
python-serial 已经是最新版 (3.4-2)。
python-setuptools 已经是最新版 (39.0.1-2)。
gperf 已经是最新版 (3.1-1)。
python-pyelftools 已经是最新版 (0.24-4)。
gcc 已经是最新版 (4:7.4.0-1ubuntu2.3)。
git 已经是最新版 (1:2.17.1-1ubuntu0.5)。
libssl-dev 已经是最新版 (1.1.1-1ubuntu2.1~18.04.5)。
python-cryptography 已经是最新版 (2.1.4-1ubuntu1.3)。
wget 已经是最新版 (1.19.4-1ubuntu2.2)。
python-pip 已经是最新版 (9.0.1-2.3~ubuntu1.18.04.1)。
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 255 个软件包未被升级。

3、执行make
[image]
会下载一些资源。
[image]

[image]

[image]
[image]

4、先运行make menuconfig
[image]

5、异常
Traceback (most recent call last):
  File "/home/sambaShare/work/esp-at-master/tools/BLEService.py", line 29, in <module>
    import yaml
ImportError: No module named yaml

[image]

6、安装 yaml
pip install PyYAML 

xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ pip install PyYAML 
Collecting PyYAML
Installing collected packages: PyYAML
Successfully installed PyYAML-5.3

xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ pip3 install PyYAML 
Collecting PyYAML
  Using cached https://files.pythonhosted.org/packages/3d/d9/ea9816aea31beeadccd03f1f8b625ecf8f645bd66744484d162d84803ce5/PyYAML-5.3.tar.gz
Building wheels for collected packages: PyYAML
  Running setup.py bdist_wheel for PyYAML ... done
  Stored in directory: /home/xwz/.cache/pip/wheels/e4/76/4d/a95b8dd7b452b69e8ed4f68b69e1b55e12c9c9624dd962b191
Successfully built PyYAML
Installing collected packages: PyYAML
Successfully installed PyYAML-5.3
[image]

ustccw, 2020-03-10:
@woshi520laoxie Please download and export your compiler firstly, it was mentioned before.

woshi520laoxie, 2020-03-10:
@ustccw 好的，我这么处理
1、删除 ~/.espressif
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ sudo rm -rf ~/.espressif
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ 

2、下载编译器并且解压 编译器
[image]

3、添加环境变量
export PATH="$PATH:/home/sambaShare/work/xtensa-esp32-elf/bin"

4、查看版本号
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/sambaShare/work/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=i686-build_pc-linux-gnu --host=i686-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-2019r2' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp-2019r2) 

5、运行make menuconfig
[image]

6、make报错
CC build/wpa_supplicant/src/wps/wps_attr_build.o
CC build/wpa_supplicant/src/wps/wps_enrollee.o
AR build/wpa_supplicant/libwpa_supplicant.a
CC build/xtensa/debug_helpers.o
CC build/xtensa/trax.o
AR build/xtensa/libxtensa.a
Generating esp32.project.ld
LD build/esp-at.elf
/home/sambaShare/work/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/sambaShare/work/esp-at-master/components/at/lib/libesp32_at_core.a: error adding symbols: no more archived files
collect2: error: ld returned 1 exit status
/home/sambaShare/work/esp-at-master/esp-idf/make/project.mk:528: recipe for target '/home/sambaShare/work/esp-at-master/build/esp-at.elf' failed
make: *** [/home/sambaShare/work/esp-at-master/build/esp-at.elf] Error 1
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$

ustccw, 2020-03-10:
@woshi520laoxie It looks a little better than that.

Maybe you could try with: `rm -rf build sdkconfig && make defconfig && make -j4`

woshi520laoxie, 2020-03-10:
@ustccw 好的，我这么操作
1、rm -rf build sdkconfig && make defconfig && make -j4
信息如下
...
ents/customized_partitions/raw_data/mqtt_key/mqtt_client.key 
AT customized partitions successfully generated: mqtt_key
generating mqtt_ca.bin: python /home/sambaShare/work/esp-at-master/tools/AtPKI.py generate_bin -b /home/sambaShare/work/esp-at-master/build/customized_partitions/mqtt_ca ca /home/sambaShare/work/esp-at-master/components/customized_partitions/raw_data/mqtt_ca/mqtt_ca.crt 
AT customized partitions successfully generated: mqtt_ca
AT customized bin generation finished
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ 

2、make
信息如下
...
AR build/wpa_supplicant/libwpa_supplicant.a
CC build/xtensa/eri.o
CC build/xtensa/debug_helpers.o
CC build/xtensa/trax.o
AS build/xtensa/debug_helpers_asm.o
AR build/xtensa/libxtensa.a
Generating esp32.project.ld
LD build/esp-at.elf
/home/sambaShare/work/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/sambaShare/work/esp-at-master/components/at/lib/libesp32_at_core.a: error adding symbols: no more archived files
collect2: error: ld returned 1 exit status
/home/sambaShare/work/esp-at-master/esp-idf/make/project.mk:528: recipe for target '/home/sambaShare/work/esp-at-master/build/esp-at.elf' failed
make: *** [/home/sambaShare/work/esp-at-master/build/esp-at.elf] Error 1

woshi520laoxie, 2020-03-12:
@xcguang 请问还有哪些尝试办法吗？

woshi520laoxie, 2020-03-12:
@ustccw 请问还有哪些尝试办法吗？

ustccw, 2020-03-12:
@woshi520laoxie It seems that the compiler type you downloaded does not match the type of your computer's operating system. and causes a link stage entry not found issue.

such as PC OS type: `64-bit`, you have to download `64-bit` compiler.

woshi520laoxie, 2020-03-12:
@ustccw 是的，我之前用的的确是32位的，我刚才改成64位的了

1、我的电脑主机是64位的
[image]

2、ubuntu系统是 64位的
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ getconf LONG_BIT
64

3、ubuntu版本信息如下
xwz@xwz-virtual-machine:/home/sambaShare/work/esp-at-master$ cat /proc/version
Linux version 5.3.0-28-generic (buildd@lcy01-amd64-009) (gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)) #30~18.04.1-Ubuntu SMP Fri Jan 17 06:14:09 UTC 2020

4.1、编译器版本:
[image]

5、查看gcc编译器版本 信息
/home/xwz/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/sambaShare/arm-linux-gcc-dir/arm-linux-gcc-gnueabihf/bin:/home/xwz/.local/bin/aos:/home/sambaShare/work/xtensa-esp32-elf/bin
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/sambaShare/work/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-2019r2' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp-2019r2) 

6、执行make menuconfig  然后执行 make -j16
make menuconfig
make -j16
编译信息:
CC build/bt/common/osi/buffer.o
CC build/bt/common/osi/allocator.o
CC build/bt/common/osi/thread.o
CC build/bt/common/btc/core/btc_task.o
CC build/bt/common/btc/core/btc_alarm.o
CC build/bt/common/btc/core/btc_manage.o
AR build/bt/libbt.a
Generating esp32.project.ld
LD build/esp-at.elf
/home/sambaShare/work/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/sambaShare/work/esp-at-master/components/at/lib/libesp32_at_core.a: error adding symbols: no more archived files
collect2: error: ld returned 1 exit status
/home/sambaShare/work/esp-at-master/esp-idf/make/project.mk:528: recipe for target '/home/sambaShare/work/esp-at-master/build/esp-at.elf' failed
make: *** [/home/sambaShare/work/esp-at-master/build/esp-at.elf] Error 1

woshi520laoxie, 2020-03-12:
@ustccw  路径一定需要放到 ~下面吗？
我是放在如下目录
[image]

woshi520laoxie, 2020-03-12:
@xcguang    ubuntu查看路径
xwz@xwz-virtual-machine:/home/sambaShare/work$ pwd
/home/sambaShare/work
xwz@xwz-virtual-machine:/home/sambaShare/work$ ls
esp-at-master      esp-idf-master.zip  xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz
esp-at-master.zip  xtensa-esp32-elf

xcguang, 2020-03-12:
你用的虚拟机吗？直接从首页下载我们安装好的吧
[image]

woshi520laoxie, 2020-03-13:
@xcguang 
1、不是，是之前下载好的麒麟18版本的
2、其实我已经有一个工程可以编译esp32_at指令了，只不过当时没有记录是用的哪个版本编译的
2.1、之前的版本用的编译器还是用的5.x的
3、现在是下载最新的esp32_at固件，然后用的编译器也是最新的

woshi520laoxie, 2020-03-13:
@xcguang 你怀疑的是ubuntu配置有问题吧，我在想如果有问题，那么我之前的esp32_at固件应该也无法编译才对了，之前可以编译的esp32_at固件，应该是去年10月份左右下载的。

xcguang, 2020-03-13:
是的，怀疑你前面使用 idf 脚本搭建新的环境导致的(搭建环境这部分, idf 里面的脚本可能还不太好用)；当前的虚拟机，我也是基于上个版本重新做的，只是更新了一下 esp32 的编译器，其它什么都没有更改；如果你有原始的虚拟机，直接下载个 esp32 编译器，将原来的替换掉应该就好了

xcguang, 2020-03-13:
不要允许

woshi520laoxie, 2020-03-13:
@xcguang 我现在已经不运行  ./install.sh    . ./export.sh  运行这个指令需要竟然esp_idf，

woshi520laoxie, 2020-03-13:
@xcguang 我现在已经不运行  ./install.sh    . ./export.sh  这种指令了，之前因为esp32_at中包含有esp_idf，是编译出编译工具链吧。
1、我之前使用的eps32编译方式也是网上下载的5.x版本的工具链，然后把ESP32_PATH的路径加入到/etc/profile 文件中    PATH=$PATH:$ESP32_PATH   之后就直接编译了，之所以运行 
./install.sh    . ./export.sh  这个指令，是因为使用esp32其他的工程。
2、如果需要重新下载你们的虚拟机可能会麻烦一些，因为我自己搭建ubutnu，还有很多其他平台的工具，会麻烦一些。我先下载你们的虚拟机吧。

woshi520laoxie, 2020-03-13:
@xcguang  你好啊
ubuntu版本 : 16版本的。
我今天换了一台ubuntu系统，这台ubuntu从来没有安装过esp32环境的。
1、我把之前下载好的esp-at-master.zip 、xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz这两个文件，放到ubuntu中
2、写了一个简单的脚本:脚本如下，执行编译，下载挺慢的...
3、我一开始使用的是新的gcc编译器，因为报错说需要用5.0的编译器，我就换成了5.0的编译器
4、
Please, commit your changes or stash them before you can switch branches.
Aborting

Toolchain path: /home/sambaShare/work/gcc-xtensa-esp32-5.x/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
The following Python requirements are not satisfied:
future>=0.15.2
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/sambaShare/work/esp-at-master/esp-idf/requirements.txt" for resolving the issue.
/home/sambaShare/work/esp-at-master/esp-idf/make/project.mk:501: recipe for target 'check_python_dependencies' failed
make: *** [check_python_dependencies] Error 1

我使用脚本执行的，脚本是
ESPAT_PROJ_PATH=$(pwd)
#GCC_PATH="/home/sambaShare/work/xtensa-esp32-elf/bin"
GCC_PATH="/home/sambaShare/work/gcc-xtensa-esp32-5.x/xtensa-esp32-elf/bin"
ESP_IDF="/home/sambaShare/work/esp-at-master/esp-idf"
export PATH=$PATH:$GCC_PATH
xtensa-esp32-elf-gcc -v
#make menuconfig
make

我安装了:sudo apt-get install git wget flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache libffi-dev libssl-dev

是否关键点是否是这个:ESP-IDF Programming Guide for setting up the required packages

JohnChenYujun, 2020-08-07:
请问这个虚拟机http://download.espressif.com/esp_share/env/ubuntu1804.ova 的帐号和密码是什么啊?

xcguang, 2020-08-24:
https://github.com/espressif/esp-at/issues/413

closed 2021-04-14 by FayeY