espressif/esp-at #359: 【紧急】AT指令设置 clientID 长度较长的字符串参数时候，无法设置成功！

sfc666, 2020-03-03:
[image]

- AT + MQTTUSERCFG在设置阿里云client ID过长返回错误和 AT + MQTTCONN设置域名时添加端口号时出现重启现象，并无法再联网。

- 复现情景：

```
\0\0
ready
AT+CWMODE=1

OK
AT+CIPSNTPCFG=1,8,"ntp1.aliyun.com"

OK
AT+MQTTUSERCFG=0,1,"robot|securemode=3,signmethod=hmacsha1,timestamp=6789|","test-02&a1Py84SEWWJ","BFDBC526D929AD5E7B077AD24BA7F79403A8CAD0",0,0,""

ERROR

```

xcguang, 2020-03-03:
对于长字节的，可以 采用 `AT+MQTTCLIENTID`  https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md#MQTT-AT 单独设置

sfc666, 2020-03-03:
这样，我测试下，还有一个问题---图片里标记2的红框，AT+MQTTCONN的第二个参数加“：任何端口号”会重起，重起后联网百分百失败，只有恢复出厂的指令才能重新联网继续！------------------ 原始邮件 ------------------

 
&mdash;

xcguang, 2020-03-03:
@ustccw 我们内部确认一下？

ustccw, 2020-03-03:
@sfc666 could you please provide the following info for us to reproduce:
- `AT+GMR` 
- `AT+SYSRAM?`
- some detailed log
- AT commands sequence

`esp-at` has not updated MQTT related functions recently, I'm not sure if it's an environmental issue or configuration issue.

sfc666, 2020-03-04:
[image]

图片中是AT + MQTTCONN的第二个参数加“：任何端口号”会重起，重起后联网百分百失败的复现

ustccw, 2020-03-04:
@aithinker-cheng It looks like the wrong MQTT command.

Please try with `AT+MQTTCONN=0,"xxx.aliyuncs.com",1883,1`.

aithinker-cheng, 2020-03-04:
[image]

  @ustccw 采用AT + MQTTCLIENTID依然返回错误，我提供阿里云相应的配置参数，你那边测试看看是否能连接得上

AT+MQTTCONN第二个参数我是故意设置成那样的，就是想说添加任意端口模组会重启并无法联网的
```

regionld: 华东2
ProductKey:    a1Py84SEWW
DeviceName:   test-02
DeviceSecret:   BvaGh4rOA62TelDTL91nrvzoMdts5zZM
连接域名:   a1Py84SEWWJ.iot-as-mqtt.cn-shanghai.aliyuncs.com
端口: 1883
client  ID:   robot|securemode=3,signmethod=hmacsha1,timestamp=6789|
username: test-02&a1Py84SEWWJ
password: BFDBC526D929AD5E7B077AD24BA7F79403A8CAD0

```

ustccw, 2020-03-04:
@aithinker-cheng please enable sys log, `AT+SYSLOG=1`

ustccw, 2020-03-05:
@aithinker-cheng please take a look.

[image]

Yuval-guo, 2020-03-06:
我这几天也在搞这个，发现AT+MQTTUSERCFG中设置阿里云clientID时的错误不是因为过长，而是因为字符串中有“，”，去掉逗号就正常，但是却不能登录阿里云了。

xcguang, 2020-03-07:
#364

xcguang, 2020-03-12:
还有问题吗？

aithinker-cheng, 2020-03-12:
ok了，感谢 @xcguang

closed 2020-03-12 by xcguang