espressif/esp-at #360: 通过HEAD请求头部数据他是可以获取到文件的名字，但是无法通过get请求去下载该文件

aithinker2222, 2020-03-04:
复现的现象具体可以查看图片，使用请求头部数据他是能获取到文件的，  但是通过get请求下载数据他是没有任何接收数据的现象
[image] 
具体指令为
AT+HTTPCLIENT=1,0,"http://update.bwtob.com:8086/bw_register/ResourceController/downloadResource?id=237-1",,,0
AT+HTTPCLIENT=2,0,"http://update.bwtob.com:8086/bw_register/ResourceController/downloadResource?id=237-1",,,1

会不会是端口号是非80的情况下无法进行get请求下载

songjiaxiu-espressif, 2020-03-04:
"http://update.bwtob.com:8086/bw_register/ResourceController/downloadResource?id=237-1" 为`Chunked` 编码, 目前不支持`Chunked` 编码的数据下载。

aithinker2222, 2020-03-05:
> “ http://update.bwtob.com:8086/bw_register/ResourceController/downloadResource?id=237-1 ”为`Chunked`编码，目前不支持`Chunked`编码的数据下载。

后续会支持这个Chunked编码的数据进行下载吗

songjiaxiu-espressif, 2020-03-06:
目前没有计划