espressif/esp-at #361: How to get SNTP time using AT Command??

thanthdinesh, 2020-03-04:
Hello I am using esp8266 module. SDK and AT version given below.

AT version:1.1.0.0(May 11 2016 18:09:56)
SDK version:1.5.4(baaeaebb)

I want SNTP time using AT Command.

I try following method to get SNTP Time, But it is not working and always get ERROR

1. SNTP configuration by:

AT+CIPSNTPCFG=1,8,"cn.ntp.org.cn","ntp.sjtu.edu.cn","us.pool.ntp.org"

ERROR

2. Get SNTP time by:

AT+CIPSNTPTIME?

ERROR

Plese let me know what might be possible reason ??

xcguang, 2020-03-04:
The firmware is too old to support this command. If you are designing a new product, recommend to use RTOS AT

thanthdinesh, 2020-03-04:
So can we can we update firmware on esp8266 ?
If yes then how can we do that ?

thanthdinesh, 2020-03-04:
I follow below links to update the firmware on esp8266 :

https://www.allaboutcircuits.com/projects/flashing-the-ESP-01-firmware-to-SDK-v2.0.0-is-easier-now/

But after follow above link and download new firmware I have getting below log with baudrate 74880 and then AT command not working.

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 816, room 16
tail 0
chksum 0xef
load 0x00000000, len 0, room 8
tail 0
chksum 0xef
load 0x00000000, len 0, room 0
tail 0
chksum 0xef
csum 0xef
csum err
ets_main.c
[image]

xcguang, 2020-03-04:
https://www.espressif.com/sites/default/files/ap/ESP8266-IDF-AT_V2.0_1.zip and you can refer to download.config in the zip. If you want run in the 1M flash, you must compile it by yourself like https://github.com/espressif/esp-at/issues/351

thanthdinesh, 2020-03-06:
I can't understand procedure given in #351 Can you explain me in details ?
If you have 1M flash binary please share with me.
Thanks,
Dinesh

thanthdinesh, 2020-03-09:
I used ESPFLASH TOOLS V2.3 and SPI Mode DOUT. Now It's Working Fine.

Thanks xcguang for helping me.

closed 2020-03-09 by thanthdinesh