espressif/esp-at #362: SSL data limit from server to esp32 (2920 Bytes)

ubichsel, 2020-03-05:
Hi Espressif Team

If we request 4 kByte of data from a server (over SSL) we receive only 2920 Bytes in one message and the rest is missing. When doing it over TCP (without SSL) it works even with 8kByte blocks. Any idea why with SSL this isn't working or which Parameters I have to tweak to increase this size?

Seems someone else had same problem: [https://www.esp32.com/viewtopic.php?t=12777](url)

Thank you in advance for your support.

ustccw, 2020-03-08:
@ubichsel Thank you for your feedback. 
We've reproduced the issue and plan to solve it this week

ubichsel, 2020-03-09:
Thanks this is great. To try it out I need issue #366 solved as well, otherways I can't use our hardware. Br

closed 2020-03-13 by igrr

ustccw, 2020-03-13:
@ubichsel Hi, please pull and try with the latest `esp-at`, it should have been fixed.