espressif/esp-at #366: Set Ethernet pin Nbr

ubichsel, 2020-03-09:
Dear Espressif team

In earlier AT-Lib versions it was possible to change the Ethernet Pins in menuconfig

[image]
Old options in Menuconfig

We have a hardware which takes advantage of this feature. With current AT-Stack I can only set RMII clock number but not the other pins (Phy Power GPIO, SMI MDC and SMI MDIO). 

[image]
Currently implemented options in Menuconfig

It would be great if you can bring back this feature because otherways we can't update our hardware with new firmware versions from you.

When starting the firmware on our hardware there is an endless restart loop because pins are incorrect. Could you please change the behaviour to start AT-Firmware properly, even the Ethernet could not be initialized (as it was in earlier AT-Versions)? Our hardware has optional Ethernet and we don't want to program different ESP-Images for the one with and without Ethernet.

We would really appreciate it if you could fix this problems. Thank you in advance.
Best regards

xcguang, 2020-03-09:
@ESP-Coco Please give some advice.

ESP-Coco, 2020-03-10:
HI @ubichsel  in IDF release/v4.0, the ethernet config has changed
[image]

ubichsel, 2020-03-10:
Thank you for your response.

Is there anything special I have to do? I completely downloaded the AT-Repository new as proposed here: [https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md](url) And all menupoints I see is
[image]

Even though I still miss the **Phy Power Pin** in your screenshot.

ESP-Coco, 2020-03-11:
@ubichsel Are you sure you are currently using the master branch?
Looking at your configuration, it seems that you are not using IDF release/v4.0,

ubichsel, 2020-03-16:
@ESP-Coco I'm not able to see the menu you have. What I do:

1. git clone --recursive https://github.com/espressif/esp-at.git
2. rm sdkconfig
3. rm -rf esp-idf
4. make defconfig
5. make menuconfig

If I have to do anything different please tell me how to do it.

ESP-Coco, 2020-03-17:
Hi @ubichsel  I am very sorry that I found the same problem with you when I used the latest master version,  IDF removed these ethernet configuration at commit 42a462d.
Now you can make changes in the main/at_eth_init.c .
```
eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();  // apply default MAC configuration
mac_config.smi_mdc_gpio_num = 23;  // alter the GPIO used for MDC signal
mac_config.smi_mdio_gpio_num = 18; // alter the GPIO used for MDIO signal
esp_eth_mac_t *mac = esp_eth_mac_new_esp32(&mac_config); // create MAC instance

eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG(); // apply default PHY configuration
phy_config.phy_addr = 1;       // alter the PHY address according to your board design
phy_config.reset_gpio_num = 5; // alter the GPIO used for PHY reset
esp_eth_phy_t *phy = esp_eth_phy_new_ip101(&phy_config); // create PHY instance
```

ubichsel, 2020-03-17:
Hi ESP-Coco
Thanks for the clarification, I'm happy you could reproduce it yourself. 
One more question. Is it still possible to set somewhere the phy Power GPIO? We have such a pin in our hardware and in eralier times it was possible to configure it.

ubichsel, 2020-03-17:
And how can I set the clock output pin? 
In menuconfig there is an option Output RMII clock from internal but which pin does it take? We have it on IO17 which was possible to set in earlier times.

ESP-Coco, 2020-03-18:
@ubichsel  
1. Poweron GPIO is a  workaround way in  previous IDF, GPIO0 is a straping pin (not only the pin of control download, but also the RMII clock). Our previous Ethernet hardware design needed to avoid the effect of GPIO0 on boot. In the new Ethernet development board, we don't need this workaround way. In Ethernet standard drive, there should be no Poweron GPIO.
In the new version of the driver, if your board needs a separate GPIO to control the power of the PHY crystal oscillator, it needs to be handled by yourself in the on_lowlevel_init_done callback function of esp_eth_config_t ( pull up or pull down this GPIO).
2. If the RMII clock mode is selected to CONFIG_ETH_RMII_CLK_OUTPUT, then GPIO0 can be used to output the REF_CLK signal. if you're not using PSRAM in your design, GPIO16 and GPIO17 are also available to output the reference clock. you can config it in `Component config -> Ethernet -> Support ESP32 internal EMAC controller`
[image]

ubichsel, 2020-06-10:
Sorry for the long break, I had to work on a different Project.

I was able to set the pins as you suggested in the previouse post. Thank you for that. With the logic analyzer I can see the clock line, mdio, mdc and power line working correctly. On console I see no error messages regarding initialization.
Unfortunately then it does noting more. As of my point of view it should check on mdio every 4 seconds (default setting) for a link on cable. This is not done, I clearly see that with my logic analyzer when sniffing MDIO and MDC pins on RMII interface (no action after init is done). Can you give me any suggestion what I can do to bring Ethernet with LAN8720 to work? Is there any AT-Command which is not in documentation where I have to activate Ethernet or something like that?

Thank you in advance.

ESP-Coco, 2020-06-12:
This should be the driver doesn't work, because esp_eth_start is not called after esp_eth_driver_install, which will start a timer to check the connection status of the network cable.

ubichsel, 2020-06-12:
Dear ESP-Coco
Thank you very much for the response. My problem is I can't debug deeper. The function esp_eth_driver_install at the end of at_eth_init is successfull but then nothing more happens. I guess this is somewhere in AT-FW which is closed source. Can you give me further instructions how to debug this error since esp_eth_driver_install was successfull (no errors printed on monitor console)?

ESP-Coco, 2020-06-15:
Hi @ubichsel  Can you replicate this problem using an Ethernet demo?
I think it is more convenient for us to analyze.

ubichsel, 2020-06-15:
Hi @ESP-Coco 
Today I got a step further. I think in the AT-Firmware the function "esp_eth_start" is not called after the "at_eth_init" function in at_eth_init.c. If I call esp_eth_start(eth_handle) in at_eth_init function at the end I get a "+ETH_GOT_IP:10.1.168.224" event when ethernet cable is connected.

The only thing missing are the following events:
- "+ETH_CONNECTED"
- "+ETH_DISCONNECTED"

Can you have a look if you miss this events in AT-Code and enable it again? Then I'm happy again with Ethernet.
Thank you again for the support.

ESP-Coco, 2020-06-16:
Hi @ubichsel Thank you for reminding me this problem. You are right. Event processing does need to be added here. This problem arises because in v4.x version, IDF do not use the old ethernet event. This needs to be adapted by AT.  I'll get back to you soon.

ubichsel, 2020-06-17:
I implemented the events in at_eth_init.c by my self and so it is working. It would be nice if you implement it into the regular AT-Stack (if others have same issue). 
For me it works right now so I close the issue.

closed 2020-06-17 by ubichsel