espressif/esp-at #368: ESP-AT command is not working

Darram007, 2020-03-11:
I followed all step and tried with 
1) github ESP-AT command steps and
2) also using flasher tools to flash bin 

I am receivng this at the terminal, `ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7004
load:0x40078000,len:14368
load:0x40080400,len:4248
entry 0x400806e0
[0;32mI (71) boot: Chip Revision: 1[0m
[0;32mI (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (39) boot: ESP-IDF v4.0-174-gf9cb434ee-dirty 2nd stage bootloader[0m
[0;32mI (39) boot: compile time 10:49:17[0m
[0;32mI (39) boot: Enabling RNG early entropy source...[0m
[0;32mI (45) boot: SPI Speed      : 40MHz[0m
[0;32mI (50) boot: SPI Mode       : DIO[0m
[0;32mI (54) boot: SPI Flash Size : 4MB[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (68) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (83) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (98) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (113) boot: End of partition table[0m
[0;32mI (118) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (125) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x29078 (168056) map[0m
[0;32mI (193) esp_image: segment 1: paddr=0x001290a0 vaddr=0x3ffbdb60 size=0x038d0 ( 14544) load[0m
[0;32mI (199) esp_image: segment 2: paddr=0x0012c978 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (201) esp_image: segment 3: paddr=0x0012cd80 vaddr=0x40080400 size=0x03290 ( 12944) load[0m
[0;32mI (214) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0x104874 (1067124) map[0m
[0;32mI (596) esp_image: segment 5: paddr=0x00234894 vaddr=0x40083690 size=0x195f8 (103928) load[0m
[0;32mI (640) esp_image: segment 6: paddr=0x0024de94 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (658) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (658) boot: Disabling RNG early entropy source...[0m
2.0.0

max tx power=78,ret=0

`
could you please have a look and let us know where the problem is ?
thank you

xcguang, 2020-03-12:
https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md

Darram007, 2020-03-12:
Thank you for the fast reply. I went through these instructions but as you can see this step i can not see in MAKE MENUCONFIG setting.I have attached tab image below. so how can configure port setting in order to activates at commands?

[image]

[image]

Darram007, 2020-03-16:
I have followed those steps but not success.
Toolchain version: esp-2019r2
ESP-IDF -esp-idf v4.0 463a9d8 

This one is a stable version so I think AT configuration process steps have to be checked and confirmed. 
#370 
 have a look on too

stoneroweast, 2020-05-08:
I experienced something similar, at the command did not work. after a few days of searching I finally found that the esp8266 module used has a capacity of 1MB. Then I do the following 3 steps:
1. change the makefile line
export ESP_AT_PROJECT_PLATFORM? = PLATFORM_ESP32
export ESP_AT_MODULE_NAME? = WROOM-32
Becomes :
export ESP_AT_PROJECT_PLATFORM? = PLATFORM_ESP8266
export ESP_AT_MODULE_NAME? = ESP8266_1MB
2. Change the file "factory_param_datas.cvs" to change the TX and RX pins:
PLATFORM_ESP8266, ESP8266_1MB, 0xfcfc, 2,2,78,0,1,13, CN, 115200,1,3, -1, -1, -1, -1
3. Change the parameters in menuconfig-> Serial flasher config-> Flash Size: 1MB

Now the AT Command is running smoothly ...

espressifelse, 2020-05-21:
These instructions will get you started with ESP-AT on Windows 10 (64 bit)

When I follow these instructions, this is the configuration I end up with
Toolchain version: esp-2019r2
Compiler version: 8.2.0
AT version:2.1.0.0-dev(b8d0d4d - May 13 2020 09:42:58)
SDK version:v4.0-327-gfe67bedee
compile time(0d586ac):May 20 2020 17:00:32
Bin version:2.0.0(WROVER-32)

Prerequisites: Python 2.7, Git, pip

1. Download the GNU Make toolchain for ESP v4.0

Go to the latest ESP documentation
Get Started » Get Started (Legacy GNU Make) » Standard Setup of Toolchain for Windows (Legacy GNU Make)
esp32_win32_msys2_environment_and_gcc8_toolchain-20191231.zip

2. Extract the toolchain to C:\

You should end up with this folder structure
C:\msys32\mingw32.exe

3. Make the esp folder

Docs » Get Started » Get Started (Legacy GNU Make) » Standard Setup of Toolchain for Windows (Legacy GNU Make)
Run mingw32.exe
mkdir -p ~/esp

4. Install ESP-AT to the esp folder

Use mingw32.exe
Navigate to this folder
C:\msys32\home\YOUR-USERNAME\esp
Run this command
git clone --recursive https://github.com/espressif/esp-at.git
You should have a folder structure like this
C:\msys32\home\YOUR-USERNAME\esp\esp-at

5. Set the IDF_PATH

Use Windows Explorer
Navigate to this folder
C:\msys32\etc\profile.d
Create a file with this name
C:\msys32\etc\profile.d\export_idf_path.sh
These are the contents of the file export_idf_path.sh
export IDF_PATH="C:/msys32/home/YOUR-USERNAME/esp/esp-at/esp-idf"
Restart mingw32.exe
Run this command
printenv IDF_PATH
The output should look like this
C:/msys32/home/YOUR-USERNAME/esp/esp-at/esp-idf

6. Install the ESP-IDF that is compatible with the ESP-AT

Use mingw32.exe
Navigate to this folder
C:\msys32\home\YOUR-USERNAME\esp\esp-at
Run this command
make menuconfig
You should have a folder structure like this
C:\msys32\home\YOUR-USERNAME\esp\esp-at\esp-idf

7. Enter your COM port

Still in menuconfig, navigate here
Serial flasher config --> Default serial port --> COM35 (for COM port 35 for example)
Here is how you can find your COM port
Plug in your ESP-32 hardware
Find the COM port it's on in your device manager

8. Enter your flash size

Still in the menuconfig
Serial flasher config --> Flash size --> 16 MB (for example)

9. Enable the Filesystem Operations (optional)

Still in the menuconfig
Component config --> AT --> 

10. Set the UART settings

Still in the menuconfig
Component config --> AT --> AT uart settings --> the uart flow control for AT port --> 0 (for no flow control)
Exit the menuconfig and save your changes when prompted

11. Set the UART port for AT commands

Use Windows Explorer
Navigate to this file
C:\msys32\home\YOUR-USERNAME\esp\esp-at\components\customized_partitions\raw_data\factory_param\factory_param_data.csv
Create a copy of the file. You should have a folder structure like this
C:\msys32\home\YOUR-USERNAME\esp\esp-at\components\customized_partitions\raw_data\factory_param\Original\factory_param_data.csv
Modify this file
C:\msys32\home\YOUR-USERNAME\esp\esp-at\components\customized_partitions\raw_data\factory_param\factory_param_data.csv
Refer to these instructions
https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md
For my hardware I want UART0 on the WROVER-32, and I use only the RX and TX lines
module_name	WROVER-32
uart_port	0
uart_tx_pin	1
uart_rx_pin	3
uart_ctx_pin	-1
uart_rts_pin	-1
tx_control_pin	-1
rx_control_pin	-1
Note that pins 1 and 3 correspond to GPIO1 and GPIO3. This translates to U0TXD and U0RXD

12. Set your target

Use Windows Explorer
Navigate to this file
C:\msys32\home\jwork\esp\esp-at\Makefile
Hake this change (if you use the WROVER-32 for example)
export ESP_AT_MODULE_NAME ?= WROVER-32

13. Install extra Python tools

Use mingw32.exe
Run this command
pip install pyyaml xlrd

14. Compile the program

Use mingw32.exe
Navigate to this folder
C:\msys32\home\YOUR-USERNAME\esp\esp-at
Run this command
make flash
Make sure your board is powered, you have the boot pin set correctly and it's been reset so the firmware will load

15. Open the serial port

Use QCOM_V1.1.exe
Find QCOM.zip here
https://github.com/espressif/esp-at/issues/153
Configure the settings: 115200 baud rate, COM35, no flow control
Open the port
Remove the boot jumper (if applicable) from your hardware
Reset your ESP32 hardware
You should see this at the end of your output
ready

16. Send your AT commands

Still using QCOM_V1.1.exe
Input string: AT
Send with Enter: Checked
Send Command
You should see this response
OK
Input string: AT+GMR
You should get a response like this
AT version:2.1.0.0-dev(b8d0d4d - May 13 2020 09:42:58)
SDK version:v4.0-327-gfe67bedee
compile time(0d586ac):May 20 2020 20:39:14
Bin version:2.0.0(WROVER-32)

techbatman29, 2020-07-18:
@espressifelse @xcguang Hello Guys,
I followed the above mentioned steps for "get you started with ESP-AT on Windows 10 (64 bit)" instructions keenly one by one,but I am struck at step 15 QCOM. 

After clicking on "open port" --> "pressed the reset button", but didn't get "Ready" as message but just left with "Open COM Port Success message"

I am also not sure about whether step 14:"make flask" was successful. I will include the response I got in the below as well as my QCOM Responses.
[image]
[image]
[image]

Please help me out & Thank you in advance

techbatman29, 2020-07-18:
https://github.com/espressif/esp-at/issues/368#issuecomment-660498489

I tried using "make flash monitor" command and it took me so long to get it done. After I again with the Step 15 Q Comm still i didn't get any affirmative response..
Please help me out & Thank you

techbatman29, 2020-07-18:
@espressifelse  @xcguang Hello Guys, 
I followed the above mentioned instruction keenly but still I am struck at Step15: QComm. I tried "Open the port" --> it was successful --> " Tried pressing reset pin" -->No expected responce.. I will include the QCOmm image please help me out and Thank you in advance..
[image]

espressifelse, 2020-07-20:
> @espressifelse @xcguang Hello Guys,
> I followed the above mentioned instruction keenly but still I am struck at Step15: QComm & Step14: Flash.
> 
> Step14: I tried to follow the ESP - AT getting started and followed but still faced a problem in make flash &
> Step 15:I tried "Open the port" --> it was successful --> " Tried pressing reset pin" -->No expected responce.. I will include the QCOmm image
> Please help me out and Thank you in advance..
> [image]

Make this change, and try again.

Send with Enter: Checked

closed 2021-04-14 by FayeY