espressif/esp-at #370: idf.py menucondif - doesn't have any AT Commands

jwestmoreland, 2020-03-12:
Hello All,

Looking at older posts - I can't seem to be able to enable the AT commands under the new tools using idy.py menuconfig.  I can't find the AT command menu like others have shown in their posts using the older msys2 process.

What's the correct sdkcondif.defaults -> sdkconfig file/process to start with?

Thanks,
John W.

ustccw, 2020-03-15:
@jwestmoreland Hi, could you please describe the issue detailly?

I am so sorry that can not get your point.

jwestmoreland, 2020-03-15:
Hello,

Please look here:
https://github.com/espressif/esp-at/issues/368

When I run menuconfig - I never see and AT Command Window as is illustrated in this:
https://github.com/espressif/esp-at/issues/368
thread.

I obviously don't have an sdkconfig that matches the latest tools for using the AT Command strings - at least what I think...

Thanks,
John W.

ustccw, 2020-03-15:
@jwestmoreland Hi, I saw your issue. It seems that there may be something wrong with your `AT port` connection.
Generally, by default, AT has two ports:
- USB0
  It is your burning port, as well as the log port of the printing system log.

- USB1
  It it your AT commands input and AT log output.
  Requires you connect the ESP32 to the computer by the fly line or serial port board.
  you could take a look [Set AT Port Pin](https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md).
  Normally, by default:
  ```
  TX ---> GPIO1  
  RX ---> GPIO3
  ```

When you burn the AT firmware, input `AT` by USB1. the `AT` will echo `OK` back.

For test whether AT has a right wire layout , you could download the AT firmware from [release](https://www.espressif.com/en/support/download/at) and have a test.

Good Luck.
CW.

jwestmoreland, 2020-03-15:
CW,

I appreciate your answer - but I think there's a misunderstanding.

My serial ports are working - but I haven't seen a factory.bin being made by the tools yet - but that's another issue.

In the thread I reference, 

[image]

That menu never comes up.  There's something incorrect I think with the sdkconfig.defaults I'm using.

Thanks,
John W.

ustccw, 2020-03-16:
@jwestmoreland Emm, are you using the latest master?
Try `git pull && rm -rf build sdkconfig && make defconfig && make menuconfig` 

Emm, is there a clean environment to compile `AT`? What i mean is that you did not change any code, even not added a directory called `at` by yourself.

Intuitively, it could be an environmental issue.

jwestmoreland, 2020-03-16:
CW,

Yes, I tried it - but for esp-at I rolled it back to the v.2.0esp32 version and for the esdp-idf back to v3.3.  I've been able to get the project to build - as I pointed out here:  https://github.com/espressif/esp-at/issues/369

but you're right.  With the new tools, Win10, Python 3.8, etc. - the build paths have got pretty mangled - I've had to just #define quite a few things to get a build to work.

I've looked at this tutorial as well:
https://youtu.be/KRyvly_SYS8
I'm going to set that up.  I tried setting up eclipse too - but there's issues with the .json files in the new environment - that's what I'm a little rusty on - I took some Java classes many year ago when they were talking about making a Java specific processor at Oracle; but I'm pretty much a C only guy even though I have taken quite a few C++ classes too - no pun intended - ha ha!

I'm working on a custom build as well - I cloned the code in main in esp-at and added a new file that basically adds an RTOS task to blink an RGB LED.  I posted this a while ago on the Arduino website since I did it originally for the MKR WiFi 1010 and the MRK Vidor 4K.  So, even though that's a simple example, it was tough getting the new build environment set up.  

It's running now - the basic AT commands are working and the RGB LEDs are blinking in a background task - but I want to enable JTAG and I would like to do that from a decent IDE - that above tutorial will be a starting place for me.  Another issue I've wondered about as well - most of those examples are using one of the Segger tools even though the ESP-Prog is supposed to be the tool of choice.  I hope there's not some reason I don't know about why those are currently available on Mouser for $12,00 each.

Thanks,
John W.

ustccw, 2020-03-17:
@jwestmoreland I guess maybe some strange problems caused by the operating system win10 environment.

Because our development tools and suites have not been verified on win10, compatibility problems or other unpredictable problems are not excluded.

I personally suggest that you could install a virtual machine on win10, such as VirtualBox, VMware,  
And then install the Ubuntu system in the virtual machine, so your problems may get better, because most of our engineer development environment is based on this, and we are easier to synchronize and reproduce problems.

Of course, if you download esp-at, and `make menuconfig`, `compile` and `run AT`  all are okey, maybe we can do something with your next issue.

jwestmoreland, 2020-03-18:
Hello CW,

I have some linux development machines I can try - but I'm trying the plugin for MS VS Code right now - getting OK results even though there's apparently some .json files that have to be checked.
Also - launching the ESP-IDF gui configuration tool doesn't appear to work for me right now - but I've got some builds to make OK.

Thanks,
John W.

ustccw, 2020-03-18:
Hi jwestmoreland,

You could have a try to launch the `make menuconfig` menu at [esp-idf](https://github.com/espressif/esp-idf).

If not work, you could raise a issue at [esp-idf issue](https://github.com/espressif/esp-idf/issues),
maybe some experienced colleagues would give you some advice.

jwestmoreland, 2020-03-18:
Hello CW,

idf.py menuconfig does come up using the terminal shell inside the VS Code environment - please see the pic:

[image]

Thanks,
John W.

ustccw, 2020-03-19:
Could I understand that `make menuconfig` can pop up the menu at `esp-idf`,
But **can not** pop up the menu at `esp-at`?

jwestmoreland, 2020-03-19:
CW,

Please take a look here and let me know if you have any questions about what I'm trying to convey.

[image]

Thanks,
John W.

ustccw, 2020-03-19:
Hi John,
Could you please navigate to `Component config`  menu, `AT command config` is under the `LWIP`.

Or you can type `/` to look for AT commands after `make menuconfig` pop up.

[image]

jwestmoreland, 2020-03-19:
Hello CW,

I will look at this as soon as I get my environment updated to debug via JTAG - as soon as I'm done I will try that.  I'm pretty sure I tried the / search method previously and there were no matches on AT commands - but I didn't try that under LWIP when I did that.

I'll try to be back with you on this in a few hrs. at the most; hopefully sooner.

Thanks,
John

jwestmoreland, 2020-03-19:
Hello CW,

Some good news - attached shows menuconfig working under VS Code; but I don't see any AT commands under LWIP - did a search too - nada.

[image]

Thanks,
John

ustccw, 2020-03-19:
Could you please drag the picture to github, for our network have no permission to access the
`user-images.githubusercontent.com`

ustccw, 2020-03-19:
Ohhh,
I thought you did it like the following:
[image]

To be honest, we have no experience on operating the `make menuconfig` at `vscode IDE`.

Sorry for that.

jwestmoreland, 2020-03-19:
CW,

Can you send me your sdkconfig and sdkconfig.defaults file you use for the above?  I don't see the AT command on my end - at least when I run menuconfig.

Thanks,
John W.

jwestmoreland, 2020-03-19:
CW,

github seems to claim all should be able to see that - I'm looking at the
'gist' stuff to see if I can get something there that's easier to access.

The only other way I know of it to make it part of a repo.

Regards,
John

ustccw, 2020-03-19:
Hi Hohn,
The following is my sdkconfig, sdkconfig.defauls is in: https://github.com/espressif/esp-at/blob/master/module_config/module_esp32_default/sdkconfig.defaults

[sdkconfig.zip](https://github.com/espressif/esp-at/files/4353200/sdkconfig.zip)

From your screenshot, I think it may be caused by the organization of your directory structure

jwestmoreland, 2020-03-19:
CW,

OK - I'll take a look and see if I can get the AT commands to load.

I've been dragging that sdkconfig around - it's the same one I'm using for the esp-at builds - and the build that I have working does work with AT commands even though I didn't config via sdkconfig - I basically went through the #defines.

Thanks,
John

ustccw, 2020-03-19:
Hi John,
  I am not sure your directory strcuture is OK or not, you could take a look at: https://docs.espressif.com/projects/esp-idf/en/v3.3/api-guides/build-system.html

If your environment still can not configurate the AT commands, could you please `zip` all of your code, I will try to make it work.

jwestmoreland, 2020-03-19:
CW,

OK - I'm trying to get this going under VS Code - let me play with this a
bit; I'll see what that can do - and then if I still can't figure it out -
I'll zip up some example code and send it to you.

Thanks,
John

jwestmoreland, 2020-03-22:
Hello CW,

I switched to Eclipse - and the sdkconfig editor seems to be working there OK - (I did get some AT Commands in the editor as well with VS Code - that entry needs to be made in sdkconfig.defaults... -) - but for now I'm going to keep running with Eclipse since the debugger is working OK under that vs. VS Code (via plugin).

Here's a screen shot:
[image]

Thanks for your help,
John

jwestmoreland, 2020-03-22:
Thanks for the help here - I think we can close this thread.

Best Regards,
John W.

closed 2020-03-22 by jwestmoreland