espressif/esp-at #372: ESP32 WROOM Module

hiflord, 2020-03-17:
I want to use ESP32 WROOM 32D wifi module. I want to control it with an stm32l0 microprocessor via uart using with AT commands. I want to know minimum pin number to communicate and which are they? Should en pin shortcut with 3.3v all the time? Also which pins should be used to update firmware inside? Do you have document for it?

hiflord, 2020-03-18:
[image]
I did all of these and I can communicate with at commands. But when I try listen module's TX0 a lot of word is coming via uart. The example picture is below.
[image]
If The communication is not stopped it the data is coming continuously. What is wrong can you help me? Is it normal I dont know but I think it is not normal.

hiflord, 2020-03-19:
When if I do IO0 pin shortcut with gnd it is stopped but I dont know if I design my pcb when exactly should I do shortcut? For firmware update IO0 connected shortcut with gnd if I want to communicate it with  AT commands IO0 is pulled of from gnd.

xcguang, 2020-03-20:
Maybe you can refer to https://www.espressif.com/sites/default/files/documentation/esp32-devkitc-v2_reference_design_0.zip

closed 2021-04-14 by FayeY