espressif/esp-at #374: ESP32 AWS Problem

hiflord, 2020-03-19:
[image]
I am doing it but AT+MQTTUSERCFG giving me always error. I am working on it over 1 month.
I have also done it with openssl commends
[image]
[image]
What is wrong can you help me please?

ustccw, 2020-03-19:
@hiflord Hi, could you please provide the error code, or more logs, by `AT+SYSLOG=1`

Emm, have you re-compiled the esp-at firmware?
To verify the validity of the certificate, please enable the SNTP before connection.
```
AT+CIPSNTPCFG=1,8,"ntp1.aliyun.com"
AT+CIPSNTPTIME?
```

If the problem can not get resolved, would you mind provide all of your certificates for us to test.

hiflord, 2020-03-19:
Hi dear @ustccw 
I tried it always before connection actually. 
[image]
this is my aws link
a2u7s0al75g2nd-ats.iot.us-east-1.amazonaws.com
[aws_changed_binfile.zip](https://github.com/espressif/esp-at/files/4353064/aws_changed_binfile.zip)
[original_aws_files.zip](https://github.com/espressif/esp-at/files/4353065/original_aws_files.zip)
I changed pem files to bin with openssl commands not python commands
Thank you.

ustccw, 2020-03-19:
@hiflord Hi, it looks work well.

[image]

From your log, maybe some wrong operations on your certficates.
Please ensure replaced the certificates by the right way again.

> error code: `0x6023` means that `AT_MQTT_CERT_LENGTH_ERROR`, all error code described here: https://blog.csdn.net/espressif/article/details/101713780

hiflord, 2020-03-19:
Did you use my bin file right? not pem files?
[image]

These are my adding adresses of bin files. What do you thing about adresses? Are they right adresses? How can I check that I added is  correct or not?

hiflord, 2020-03-19:
I fixed it. I dont know how it is happened but I think The process of pem file to bin file I was using openssl commands but now I used python commands using with terminal. After that process it is fixed. Thank you <3

closed 2020-03-19 by hiflord