espressif/esp-at #376: esp-at on ESP32DOWQ5

gipobbe, 2020-03-19:
Hi everyone, 
I'm having difficulties trying to build a firmware to flash on the board above. I've read about compatibility issues between esp-idf, esp-at and toolchain (cmake on idf.py and make), and I think this is my issue.
I'm trying to build the project on Ubuntu 18. 
While building the firmware I just keep getting errors over errors which seems pretty impossible to resolve. Some example: 

***
Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).
***

esp_at_port_recv_data_notify' collect2: error: ld returned 1 exit status /esp32/esp32-at/example/at_http_webserver/../../esp-idf/make/project.mk:457: recipe for target '/esp32/esp32-at/example/at_http_webserver/build/esp-http-at.elf' failed make: *** [/esp32/esp32-at/example/at_http_webserver/build/esp-http-at.elf] Error 1

***

And the last, the only one on which I took track of the various versions of the components: 
esp-idf 4.0.0, esp-at 2.0 and xtensa 2019r2 (prior versions doesn't work)

The error is:

/home/giovanni/esp/esp32-at/esp-idf/make/component_wrapper.mk:291: recipe for target 'asio/asio/src/asio.o' failed
make[1]: *** [asio/asio/src/asio.o] Error 1
/home/giovanni/esp/esp32-at/esp-idf/make/project.mk:590: recipe for target 'component-asio-build' failed
make: *** [component-asio-build] Error 2
*****

To keep it short: there's a combination of esp-idf, esp-at and xtensa which just work on ubuntu?
PS: why esp-at don't have its own esp-idf inside even using the recursive option from git? It can save some problems like mine about compatibility..

ustccw, 2020-03-20:
@gipobbe Hi, `esp-idf` and `xtensa` can work on Linux, Windows, Mac OS platform. https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html#step-1-install-prerequisites

`esp-at` will clone `esp-idf` and its submodules when user `make`, according to `module_config/module_xxx/IDF_VERSION`

gipobbe, 2020-03-20:
> `esp-at` will clone `esp-idf` and its submodules when user `make`
Yess, but nothing change. It still don't work. Even because esp-at 2.0 download esp-idf 4.0.0 which is not compatible with. (Gmake in idf.py doesn't work). So, if I use esp-at 2.0 which version of esp-idf i should use? I tried 3.3 but it didn't work..

gipobbe, 2020-03-20:
Solved. My problem at building projects came from the fact that I missed this step while configuring the sdk:
step2:[install pip](https://pip.pypa.io/en/latest/installing/) 

Now I can build esp32-at firmware succesfully. Thanks for the help ustccw, but the problem was on my end.

closed 2020-03-20 by gipobbe