espressif/esp-at #378: REQ and SUGGESTION - Complete sdkconfig(.defaults) and at_customize.bin instructions

jwestmoreland, 2020-03-25:
Hello All:

First - a request:
May I get a complete sdkconfig.defaults (and possibly sdkconfig) that has all of the ESP-AT features in the file plus building the at_customize.bin file?

Second - a suggestion:
Also - in the instructions on how to build the .bin - I believe the offset isn't included - for an offset of 0x20000:
--offset 0x20000

should be added to the command line for building at_customize.bin.
 => https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Customize_Partitions.md

Thanks In Advance,
John W.

xcguang, 2020-03-29:
You can enable all configurations about AT in `Component config > AT`.

jwestmoreland, 2020-03-30:
Hello @xcguang ,

On some projects when I use sdkconfig - that doesn't even show if there's no entry in the sdkconfig.defaults file.

AFAIK, menuconfig is deprecated for v4.0 and above, correct?

Thanks,
John

xcguang, 2020-03-30:
Firstly, run
```
export ESP_AT_PROJECT_PLATFORM=PLATFORM_ESP32
export ESP_AT_MODULE_NAME=WROOM-32
export ESP_AT_PROJECT_PATH=$(pwd)
```

then,
```
./esp-idf/tools/idf.py build
```

jwestmoreland, 2020-03-30:
Hello @xcguang,

Thanks - that gives the sdkconfig equivalent for the current project - but I was hoping to get an sdkconfig.defaults file that had all of the AT options in it for reference.

Regards,
John

closed 2020-04-23 by jwestmoreland