espressif/esp-at #379: SPI instead of UART

yldzmuhammed, 2020-03-28:
Hello everyone,

I used so many project esp32 and its great. Thanks all of you guys for this great effort.

Now i wonder if i can use esp32 at with some high speed bus like spi instead of uart.

In my project, i am getting raw camera data and i am processing it with 200mhz mcu. And i want to stream it with wifi. Maybe video server or somethink else. And also, i need to use ble on the esp32.

What should i do to get it done?

Is it possible to stream video signal from uart and use ble? Or should/can i change the bus to spi instead of uart?

xcguang, 2020-03-29:
I think, maybe the SDIO is a better way

yldzmuhammed, 2020-03-29:
@xcguang thanks for answer. Could you give me more explanation?

Is there any example for that? Or ready to use firmware/hex?

xcguang, 2020-03-29:
You can select SDIO or SPI in `Component config > AT > Enable AT > communicate method for AT command`. And there is a host demo for ESP32 SDIO AT `examples/at_sdio_host`, maybe it is helpful to you

yldzmuhammed, 2020-03-29:
I checked but not clear for now. I will deeply look at it. Thank you.

yldzmuhammed, 2020-03-30:
hi again.

I looked it up. But somethings not clear to me.

If i follow your steps, can i access esp32 with spi from my spi master mcu?

For example we can send AT+CWJAP="mySSID","mypassword"#013#010 command via uart.

How can i do the same job over spi? Is it same, should i send same command from spi?

xcguang, 2020-03-30:
@ESP-Coco Please give some suggestion.

ESP-Coco, 2020-03-30:
@yldzmuhammed Hi， you are right. Just send the same command on the upper layer, and the system will be packaged as SDIO command and sent through SPI. you can refer to this document : https://github.com/espressif/esp-at/tree/master/examples/at_sdspi_host ,

yldzmuhammed, 2020-03-30:
@ESP-Coco thanks.

In that case, esp32 will be slave an my mcu (whatever i use) will be master so that i can send at commands to the esp32. Correct?

If yes, which pin of the esp32 should i use? I am a little confused..

ESP-Coco, 2020-03-30:
@yldzmuhammed You are right, ESP32 will be SDIO slave, the pins ESP32 uses are shown in the ESP32 datasheet.
[image]
In ESP32's development release, these pins will be labeled D0, D1, CMD and so on.

closed 2021-01-18 by xcguang