espressif/esp-at #381: AT_HTTP_WEBSERVER - Help with mkfatfs

jwestmoreland, 2020-03-30:
Hello All,

I've made the example and the web-page interface is running - but I'm not able to mount the example file system properly.

I built the mkfatfs.exe for win10 - and it does make the image file - but something must be wrong since I can't mount it from the app -

[image]

I've had to shrink down some things to fit into 2MB - but the partition table seems to be OK unless I'm missing something.

Thanks In Advance,
John W.

ESP-Coco, 2020-04-01:
Hi @jwestmoreland Can you tell me which version of AT you are using?  Mkfatfs is an open source library . Could you please use this demo(URL: https://github.com/jkearins/ESP32_mkfatfs ) to test it.

jwestmoreland, 2020-04-01:
Hello @ESP-Coco ,

I'm using the ESP-AT master as posted on 3/26/2020.

And, I'm using the mkfatfs as posted in the AT_HTTP_WEBSERVER demo - and I attached that to my post.

I can try looking at what you've suggested in another repository - but was hoping someone that has used this demo before successfully could take a look at the posted binary and tell me if the fat_fs bin produced is correct or not.

The AT_HTTP_WEBSERVER demo works except for mounting the fat flash bin made by the executable.

Thanks,
John

jwestmoreland, 2020-04-01:
Hello @ESP-Coco ,

I was able to run the demo you requested - I used the binary I made to write the flash contents:

I (2088) cpu_start:
<>*** (2)available heap: 0x47730 ***<>

I (2193) FATFS: Mounting FAT partition...
I (2194) gpio: GPIO[25]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2197) wl_flash: updateV1_V2 Update from V1 to V2, crc=0xe67fe48d,
I (2198) gpio: GPIO[26]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2211) gpio: GPIO[27]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2292) wl_flash: updateV1_V2 max_pos=237, pos=186, state.ver=1, state2.ver=1
I (2735) FATFS: Done.
I (2736) FATFS: ============= Begin of root contents
I (2739) FATFS:         ESP32Explorer.html                         8657 bytes
I (2740) FATFS:         ESP32Explorer.js                          11854 bytes
I (2743) FATFS:         <images>
I (2745) FATFS:         <jquery>
I (2748) FATFS:         <jQuery-File-Upload-9.18.0>
I (2751) FATFS:         <jqueryui>
I (2754) FATFS:         <jstree>
I (2756) FATFS: ============= End of root contents

Regards,
John W.

ESP-Coco, 2020-04-02:
@jwestmoreland  I also found some problems when compiling at_http_webserver demo on the master branch. It seems that this demo is not compatible with the latest version . I will fix this problem and then synchronize the results to you in time.

jwestmoreland, 2020-04-02:
Hello @ESP-Coco ,

Thanks in advance for doing that - I think that will help a lot of us trying to get this to work with IDF v4.x+ and sdkconfig and all.

I did get the code to mount using the vfs option that's commented out in the code but it still doesn't read the filesystem made with mkfatfs.

[image]

Thanks Again,
John W.

jwestmoreland, 2020-04-02:
Hello Again @ESP-Coco ,

I wrote the fat filesystem again a little differently - and am now getting this:

[image]

I'm not sure why the Chinese font is picked since I thought I had checked that and the US English font should be selected - but I suppose this is progress.

My target has 2MB instead of 4MB so I've had to squeeze this down a bit - but at least that works.

Note I haven't been able to get the project in Eclipse to kick off the fat fs build(yet) - hope you can resolve that.

Thanks,
John W.

ESP-Coco, 2020-04-02:
@jwestmoreland I have found the reason why fatfs cannot be mounted, you need to modify the configuration(Modify CONFIG_SPI_FLASH_USE_LEGACY_IMPL = y, or select `Component config`-> `SPI Flash driver`-> `Use the legacy implementation before IDF v4.0` ) item in sdkconfig and then recompile the project.
When you compile the at_http_webserver demo in the esp-at/examples directory, the system automatically compiles fatfs and then generates the fatfs image at 0x7000, then you just need to execute `make flash` like any other program to burn it to ESP32.
The chinense font looks like CSS material appeared problems, this is the demo we provided, just for reference.

jwestmoreland, 2020-04-02:
@ESP-Coco ,

That's for loading and flashing the program on the ESP32 - writing the file system is orthogonal to that I believe.

If that was the problem - I would not have been able to post what I had in the original post in this thread.

Regards,
John W.

jwestmoreland, 2020-04-04:
Hello Again @ESP-Coco and Espressif,

I thought we could pass an AT command via HTML and this example would process that.

I guess to implement the full AT suite, would require extending

HttpdBuiltInUrl builtInUrls[] = { ... }; 

correct?  Or am I missing something?

Thanks In Advance,
John W.

ESP-Coco, 2020-04-07:
Hi @jwestmoreland , you are right. If you need to extend the AT command, first you need to implement the HTML file yourself, and then according th path to  process in the HttpdBuiltInUrl, including reading the parameters on the web page, and then call the corresponding AT command to execute.

jwestmoreland, 2020-04-07:
Hello @ESP-Coco,

OK - thanks for clarifying that for me.

Best Regards,
John

jwestmoreland, 2020-04-16:
Hello All,

It's OK to close this I guess.

Thanks,
John W.

closed 2020-04-16 by jwestmoreland