espressif/esp-at #382: 请求帮助:ESP32的BLE发现距离能否缩小

woshi520laoxie, 2020-04-01:
请问大家，ESP32的BLE发现距离能否缩小
>由于场景的需求，蓝牙过多的时候可能导致多个设备同时搜索到，期望能把蓝牙发现的距离缩小，请问这个能否进行配置。蓝牙可发现，可连接距离限制到 3M以内。

xcguang, 2020-04-01:
可以考虑 AT+RFPOWER 来降低 BLE 功率

woshi520laoxie, 2020-04-01:
@xcguang  是的，我配置了最低，AT+RFPOWER=0,7,7,7   这个时候蓝牙理论上功率是最低了，但是距离还是很远，起码10米以上了，期望能达到蓝牙搜索距离2米以内就无法搜索到就最好了。有办法吗

woshi520laoxie, 2020-04-01:
@xcguang 
你好啊，我另外问一个问题，目前ESP32Flash里面的保存的ssid，pwd能否读取出来
目前如果没有AT指令可以读，那么这个ssid和pwd是保存在哪个位置的
如果我需要使用，是否需要开启文件系统。获取分区表的信息。或者自行修改固件呢？

xcguang, 2020-04-01:
> 你好啊，我另外问一个问题，目前ESP32Flash里面的保存的ssid，pwd能否读取出来
目前如果没有AT指令可以读，那么这个ssid和pwd是保存在哪个位置的
如果我需要使用，是否需要开启文件系统。获取分区表的信息。或者自行修改固件呢？

系统上电自动读取

xiewenxiang, 2020-04-01:
目前发射功率没法再调的更低了

woshi520laoxie, 2020-04-01:
@xcguang 有指令可以获取吗？
是这样的，我需要对比一下，我将要设置的  ssid 和pwd和之前配置的是否一致。

xcguang, 2020-04-01:
密码获取不到，AT+CWJAP? 可以获取 SSID

woshi520laoxie, 2020-04-01:
@xcguang 如果我需要获取模组保存到flash里面的 wifi 密码，有那种办法吗？

xcguang, 2020-04-01:
没有办法

woshi520laoxie, 2020-05-12:
@xcguang 因为我想要看看用户设置的密码和之前设置的密码是否一致，如果wifi一直连接不上的情况下，在某些情况下可以判断是否要重新连接wifi

xcguang, 2020-05-12:
涉及到隐私，不能获取

woshi520laoxie, 2020-05-12:
@xcguang 我还提了另外一个蓝牙相关的问题，手机很难和ESP32的ble建立连接，可以帮我看看吗？搜索《蓝牙》关键字可以看到。

closed 2020-08-10 by xiewenxiang