espressif/esp-at #386: Unable to connect SSL even after flashing the Root CA certificate

tohyf, 2020-04-05:
I flashed my ESP32-WROOM with the ESP_AT_V1.2 obtained from the Espressif official download page. It works fine for connecting to normal TCP connections.
However, when I want to connect to HTTPS (SSL), it doesn't work.

I tried downloading the root CA certificate (in this example I try google.com's root CA) and generate the bin file:
`python AtPKI.py generate_bin -b  esp32_google_root_ca.bin ca ~/Downloads/google_globalsign_root_ca.pem`

Then, I flashed the bin file to offset 0x2e000 (the client CA offset):
`python esptool-master/esptool.py --port /dev/ttyUSB0 write_flash 0x2e000 ~/esp32_google_root_ca.bin`

It still doesn't work:
```
[23:17:38:042] AT+CIPSSLCCONF=2,0,0
[23:17:38:054] 
[23:17:38:056] OK
[23:17:39:929] AT+CIPSTART="SSL","google.com",443
[23:17:40:103] 
[23:17:40:108] ERROR
```
Did I miss anything?

ustccw, 2020-04-08:
@tohyf Please try to instead of `components/customized_partitions/raw_data/client_ca/client_ca_00.crt` with your google root CA.
compile `esp-at` and test again.

tohyf, 2020-04-08:
Ok, let me try this. I can place multiple of root CA here, right? How to 
do it?

tohyf, 2020-04-08:
> @tohyf Please try to instead of `components/customized_partitions/raw_data/client_ca/client_ca_00.crt` with your google root CA.
> compile `esp-at` and test again.

I tried your suggestion, setting up the toolchain, compiling the esp-at using "make clean" and then "make flash", but still same error. I have attached my client_ca_00.crt file for your reference.
[client_ca_00.zip](https://github.com/espressif/esp-at/files/4450706/client_ca_00.zip)

tohyf, 2020-04-08:
Ok, seems like I forgot to set the NTP server settings. Once I enable the server by doing : AT+CIPSNTPCFG=1,8 , I am able to connect.
Also, it seems like with the new compiled version of ESP-AT, and the new esptool in the esp-idf, I can directly generate and write into the offset 0x2e000 and it still works.

ustccw, 2020-04-13:
@tohyf `0x2e000` is used for `client ca` storage. You can instead of it with right bin.

About new certificate connect error, I am not sure had you set `AT+SSLCCONF`.

Onemore, if some problem appear, please providing the detailed log, and all the steps.  
thanks.

tohyf, 2020-04-16:
basically I directly flash the ESP32 using the official bin of ESP_AT_V1.2 , then i generate the root CA bin file using the AtPKI.py tool, then flash it to the offset 0x2e000.
The commands I use to connect are these:

AT+CIPSNTPCFG=1,8
AT+CIPSSLCCONF=2,0,0
AT+CIPSTART="SSL","google.com",443

I found that flashing the bin to the 0x2e000 only works once or twice(even with the ESP AT that I compile myself), and after that, any SSL connection will just throw error. I would need to flash everything again from the compiled version again to make it work.
What detailed log do you need?