espressif/esp-at #387: Task watchdog got triggered.

lakshmisivareddy, 2020-04-07:
Hi 
i implemented custom AT command for connecting to AWS IOT with AWS SDK,
But After calling aws_iot_mqtt_connect watching dog getting triggered    
i tried creating aws_iot_mqtt_connect  in a sup rate Task as well 
as ways ended up
task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

Any Help Would Be Appreciated

Reagrds
Siva

ustccw, 2020-04-08:
@lakshmisivareddy Hi, i am not sure whether you connect to aws by AT+MQTT commands.
here is a guide about to connect to aws by esp-at, [click](https://blog.csdn.net/espressif/article/details/105045610).

Onemore, task watchdog can be disabled at `make menuconfig`, you could try to type `/` to search.
task watchdog caused by TLS connect time over the task wdt timeout, it is just a warning.

lakshmisivareddy, 2020-04-08:
Hi ustccw Thanks For the response,
i am  implementing aws iot stack in At custom 

created a custom At function 
{ "+UART_PUT", NULL, NULL, at_setupCmdPut,NULL } };

static uint8_t at_setupCmdPut(uint8_t para_num) {
 //rading the data 
 // check the aws connection 
// if not connected connect to aws 
   connect_aws();
// send message
  publish();

}

int connect_aws()
{
    const size_t stack_size = 15*1024;
	
    
    strcpy(AWS_IOT_HOST_ADDRESS,hostAddress);
    IoT_Error_t rc = FAILURE;

    IoT_Client_Init_Params mqttInitParams = iotClientInitParamsDefault;
    IoT_Client_Connect_Params connectParams = iotClientConnectParamsDefault;
    

    ESP_LOGI(TAG, "AWS IoT SDK Version %d.%d.%d-%s", VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH, VERSION_TAG);

    mqttInitParams.enableAutoReconnect = true;  //Auto Reconnect
    mqttInitParams.pHostURL = AWS_IOT_HOST_ADDRESS;
    mqttInitParams.port = 8883;

    mqttInitParams.pRootCALocation = (const char *)aws_root_ca_pem;
    mqttInitParams.pDeviceCertLocation = (const char *)certificate_pem_crt;
    mqttInitParams.pDevicePrivateKeyLocation = (const char *)private_pem_key;

    mqttInitParams.mqttCommandTimeout_ms = 20000;
    mqttInitParams.tlsHandshakeTimeout_ms = 5000;
    mqttInitParams.isSSLHostnameVerify = true;
    mqttInitParams.disconnectHandler = disconnectCallbackHandler;
    mqttInitParams.disconnectHandlerData = NULL;

    rc = aws_iot_mqtt_init(&client, &mqttInitParams);
   
    if(SUCCESS != rc) {
        ESP_LOGE(TAG, "aws_iot_mqtt_init returned error : %d ", rc);
        return rc; 
    }
    connectParams.keepAliveIntervalInSec = 10;
    connectParams.isCleanSession = true;
    connectParams.MQTTVersion = MQTT_3_1_1;
    connectParams.pClientID = clientID;
    connectParams.clientIDLen = (uint16_t) strlen(clientID);
    connectParams.isWillMsgPresent = false;

    ESP_LOGI(TAG, "Connecting to AWS...");
    
    do {
		
		printf("aws_iot_mqtt_connect \r\n");
        rc = aws_iot_mqtt_connect(&client, &connectParams);
        
        if(SUCCESS != rc) {
            ESP_LOGE(TAG, "Error(%d) connecting to %s:%d, \n\rTrying to reconnect", rc, mqttInitParams.pHostURL, mqttInitParams.port);
            
        }
		vTaskDelay(1000 / portTICK_RATE_MS);
        
    } while(SUCCESS != rc);
	
	//Auto Reconnect
    rc = aws_iot_mqtt_autoreconnect_set_status(&client, true);
    if(SUCCESS != rc) {
        ESP_LOGE(TAG, "Unable to set Auto Reconnect to true - %d", rc);
    }    
    
    if(rc == SUCCESS)
	{
	    xTaskCreate(&UTEC_OEM_AWS_IoT_Task, "UTEC_OEM_AWS_IoT_Task", stack_size, NULL, 5, NULL);
	}

    return rc;
}
 
after hitting the  aws_iot_mqtt_connect wdt warning is coming 
task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

how do i fix this ?

ustccw, 2020-04-13:
@lakshmisivareddy 
I am not sure if you're using `ESP32 chip` or `ESP8266 chip`.
Onemore, it looks like `UTEC_OEM_AWS_IoT_Task` had a very large stack size(15KB), normally, task stack could be set to below 8KB, except for necessary.

About task watchdog, you could change the timeout internal at `make menuconfig`.  
In fact, this warning does not affect the function.

closed 2020-08-18 by xcguang