espressif/esp-at #388: ESP_AT 指令 AT+CWJAP= 和 AT+CIPSTART= 无法连接的时候总是busy

woshi520laoxie, 2020-04-10:
由于发送指令
1、连接热点:
AT+CWJAP=
2、建立TCP连接
AT+CIPSTART=
如果无法连接，这个时候发送其他指令一直回复 busy p... 导致无法做其他的事情

例子:
发送数据 b'AT+CWJAP="xxx","xxx@SINO789"\r\n' 
接收:35:06.111961
AT+CWJAP="xxx","xxx@SINO789"

发送数据 b'AT\r\n' 
接收:35:07.966897

busy p...
发送数据 b'AT\r\n' 
接收:35:08.766820

busy p...
发送数据 b'AT\r\n' 
接收:35:09.566515

busy p...
发送数据 b'AT\r\n' 
接收:35:10.318283

busy p...
发送数据 b'AT\r\n' 
接收:35:11.017251

busy p...
发送数据 b'AT\r\n' 
接收:35:11.780750

busy p...
发送数据 b'AT\r\n' 
接收:35:12.530539

busy p...
发送数据 b'AT\r\n' 
接收:35:13.294863

busy p...
发送数据 b'AT\r\n' 
接收:35:14.050429

busy p...
发送数据 b'AT\r\n' 
接收:35:14.779692

busy p...
发送数据 b'AT\r\n' 
接收:35:15.567486

busy p...
发送数据 b'AT\r\n' 
接收:35:16.253616

busy p...

发送数据 b'AT\r\n' 
接收:35:16.983464

busy p...

发送数据 b'AT\r\n' 
接收:35:17.694731

busy p...

发送数据 b'AT\r\n' 
接收:35:18.414301

busy p...

发送数据 b'AT\r\n' 
接收:35:19.117954

busy p...

发送数据 b'AT\r\n' 
接收:35:19.812736

busy p...

发送数据 b'AT\r\n' 
接收:35:20.499614

busy p...

发送数据 b'AT\r\n' 
接收:35:21.233527
+CWJAP:4

ERROR
AT

OK 

建立TCP连接  由于我们的服务器，所以就不写出来了，目前wifi已经连接
发送数据 b'AT+CIPSTART=1,"TCP","xxxx",xxx\r\n' 
接收:AT+CIPSTART=1,"TCP","xxxx",xxx

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
busy p...

发送数据 b'AT\r\n' 
接收:
ERROR
AT

OK 

发送数据 b'AT\r\n' 
接收:AT

OK

woshi520laoxie, 2020-04-13:
请问这个问题，有坛友知道的吗？

jwestmoreland, 2020-04-16:
这是用于WiFi还是以太网？

woshi520laoxie, 2020-04-18:
用于wifi，esp32的wifi  at指令

StevenRoan, 2020-06-04:
請問這問題有後續更新嗎？ 這邊也卡到。

ustccw, 2020-06-04:
@all please update to latest version, we fixed a semaphore synchronization issue before.

In addition, please providing the steps, logs, at command sequences and `AT+GMR` information for us to debug if it still exits.
thanks

woshi520laoxie, 2020-06-04:
发送下一条指令，需要等待比较长的时间。

你好啊，我测试的指令是这样的。版本号如下接收:AT+GMR
AT version:2.1.0.0-dev(3e80826 - Mar 13 2020 03:32:45)
SDK version:v4.0-beta2-174-g99fb9a3f7-dirty
compile time():Jun 2 2020 16:44:18
Bin version:2.0.0(WROOM-32)

OK 

执行的错有指令如下

发送数据 b'AT+RST\r\n' 
 
接收:AT+RST

OK 

 
接收:
ready 

 
发送数据 b'AT+CIPMUX=1\r\n' 
 
接收:AT+CIPMUX=1

OK 

 
发送数据 b'AT+CWMODE=3\r\n' 
 
接收:AT+CWMODE=3

OK 

 
发送数据 b'AT+CWMODE=?\r\n' 
 
接收:AT+CWMODE=?
+CWMODE:(1-3)

OK 

 
发送数据 b'AT+CWJAP="SINO1111","PILOT@SINO"\r\n' 
 
接收:AT+CWJAP="SINO1111","PILOT@SINO"

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
发送数据 b'AT+PING="14.29.145.35"\r\n' 
 
接收:
busy p...

 
接收:+CWJAP:3

ERROR&nbsp;

woshi520laoxie, 2020-06-04:
@ustccw 上面是我的一些操作的log，我的这个版本是否还是旧的版本，基本上操作指令都在上面了。

woshi520laoxie, 2020-06-04:
@ustccw 在连接ap的时候，发送AT\r\n指令的时候也是 一直都是busy p...的

xcguang, 2020-06-04:
所有命令是串行执行，每条命令都会返回 OK 或者 ERROR。CWJAP 耗时较久，请等待 AT+CWJAP 返回执行结果后再发下一天命令

woshi520laoxie, 2020-06-08:
@xcguang 但是这个时间特别的久，需要等到10多秒，这个时候还需要使用蓝牙，但是由于wifi 的热点ap断开了，连接不上，会一直连接，这个时候蓝牙就无法使用了。同样，连接wifi热点和建立TCP连接都比较耗时。这个时候一直都是busy，挺影响体验的。如果时间短一些会好些。

FayeY, 2021-04-14:
由于这个问题较久没有更新了，暂时关闭。如果问题在最新版本 esp-at 仍然存在，欢迎 reopen issue 再次开启讨论。谢谢！

closed 2021-04-14 by FayeY