espressif/esp-at #390: esp-at build problem

Honzik321, 2020-04-11:
Hello,
I have some issue at components/customized_partitions/CMakeLists.txt:

CMake Error at components/customized_partitions/CMakeLists.txt:20 (esptool_py_flash_project_args):
  Unknown CMake command "esptool_py_flash_project_args".

How to fix it?
Thank you, 
Jan

Honzik321, 2020-04-12:
Hello, I have installed stable version of esp-idf (v4.0) where esptool_py_flash_project_args function is included. Last error during building remains (see below).

[1077/1078] Running utility command for customized_bin
FAILED: esp-idf/customized_partitions/CMakeFiles/customized_bin.util
cmd.exe /C "cd /D C:\Users\dell\esp\esp-at\build\esp-idf\customized_partitions && mkdir -p C:/Users/dell/esp/esp-at/build/include && cp C:/Users/dell/esp/esp-at/build/config/sdkconfig.h C:/Users/dell/esp/esp-at/build/include/sdkconfig.h && C:\Users\dell\.espressif\python_env\idf4.0_py2.7_env\Scripts\python.exe C:/Users/dell/esp/esp-at/components/customized_partitions/raw_data/at_customized_partition_gen.py --tools_dir C:/Users/dell/esp/esp-at/components/customized_partitions/generation_tools --output_dir C:/Users/dell/esp/esp-at/build/customized_partitions --flash_args_file C:/Users/dell/esp/esp-at/build/flash_customized_partition_args"
The syntax of the command is incorrect.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

My configuration:
esp-idf: v4.0
Python 2.7.17
git version 2.21.0.windows.1
esp-at: v2.0.0.0_esp32-161-gbd8039b

Please help me, I'm, trying to build esp-at for couple of days without success.

Thank you,
Honzik

ustccw, 2020-04-13:
@Honzik321 Please according to [ESP-AT Get Started](https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md)

Honzik321, 2020-04-13:
Hi,
I followed the ESP-AT Get Started. Only one improvisation had to be done in file CMakeLists.txt, because, "sed" command doesn't work on my Windows 10 => release, commit, repository was manually set according to \esp-at\module_config\module_esp32_default\IDF_VERISON. I have also added Environment variables:

ESP_AT_MODULE_NAME = "WROOM-32"
ESP_AT_PROJECT_PATH = "C:\Users\dell\esp\esp-at"
ESP_AT_PROJECT_PLATFORM = "PLATFORM_ESP32"

CMakeLists.txt changes:
string(REPLACE "branch:" "" esp_at_idf_branch "release/v4.0")
string(REPLACE "commit:" "" esp_at_idf_commit "f9cb434eeb90be4f34893a278d0a70a8b21e4ca2")
string(REPLACE "repository:" "" esp_at_idf_repository "https://github.com/espressif/esp-idf.git")

Python 2.7.17 - OK 
pip install pyyaml xlrd - OK
git clone --recursive https://github.com/espressif/esp-at.git - OK
idf.py menuconfig - OK WORKS

BUILD => idf.py build => STILL ERROR IN GENERATING customized_partitions BIN FILES
idf.py build => STILL ERROR IN STEP  [1077/1078]

[1077/1078] Running utility command for customized_bin
FAILED: esp-idf/customized_partitions/CMakeFiles/customized_bin.util
cmd.exe /C "cd /D C:\Users\dell\esp\esp-at\build\esp-idf\customized_partitions && mkdir -p C:/Users/dell/esp/esp-at/build/include && cp C:/Users/dell/esp/esp-at/build/config/sdkconfig.h C:/Users/dell/esp/esp-at/build/include/sdkconfig.h && C:\Users\dell\.espressif\python_env\idf4.0_py2.7_env\Scripts\python.exe C:/Users/dell/esp/esp-at/components/customized_partitions/raw_data/at_customized_partition_gen.py --tools_dir C:/Users/dell/esp/esp-at/components/customized_partitions/generation_tools --output_dir C:/Users/dell/esp/esp-at/build/customized_partitions --flash_args_file C:/Users/dell/esp/esp-at/build/flash_customized_partition_args"
The syntax of the command is incorrect.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Thank you for your help,
Honzik

ustccw, 2020-04-14:
you can do something on the error
```
The syntax of the command is incorrect.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Honzik321, 2020-04-14:
I decided to change from windows to Linux. Probably it is not possible to build on windows. I'm confused. I'm new in esp-idf. The esp-idf itself is possible to build on windows OS.

jwestmoreland, 2020-04-16:
@Honzik321 ,

It is possible - but it can take a little work - some quick hints:

To just get the ESP32 stuff to build - disable (remove from any PATH(S)) - ANY other tools - cygwin, msys, etc - or you'll regret it.

I got stuff finally running under Eclipse and Windows 'Code' - there are plug ins for both of those - I 
have JTAG running under Eclipse...

First, install the ESP-IDF - the latest AT lib goes to ESP-IDF v4.0 - make sure you can compile and download and run some of the simple examples the IDF has.

Repeat that process for the AT lib - then move on to your application specific stuff.

There's a lot of places this can crater - some are code incompatibilities - some are not enabling the 'legacy' SPI FLASH code if your target requires that.  

I have a somewhat custom board and the presto-magic install script thing that runs and updates the known universe doesn't work for me.

Make sure you can run idf.py menuconfig from your command line.  If sdkconfig.defaults is incorrect and/or ..\build\config\sdkconfig.h is incorrect - you're sunk.  Make sure those are OK.

If using Eclipse - make sure you can kick off the sdkconfig tool (runs JSON I think) - it'll croak if all is not well - can you can bet your build is sure to fail if that doesn't come up.  Something similar will happen under Windows 'Code' thing too.

Well, YMMV.
HTH,
John W.

xcguang, 2020-04-17:
I suggest, if possible, you'd better use virtual machine on windows OS, the image can be downloaded from http://download.espressif.com/esp_share/env/ubuntu1804.ova, in which we have deployed compiling environment.

Honzik321, 2020-04-17:
Thank you, I have already built esp-at on my Rapberry pi (Linux Raspbian) with success.  I will try virtual machine, but now I have trouble with at sleep command and uart hardware flow control (rts,cts). See my next topic here.

closed 2020-04-17 by Honzik321