espressif/esp-at #394: esp-at make error with ESP8266_RTOS_SDK

pablinza, 2020-04-15:
Hello. 
A cannot build esp-at firmware to ESP8266
ESP8266_RTOS_SDK and xtensa toolchains are working 

On esp-at directory Makefile option was changed to:
export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP8266
export ESP_AT_MODULE_NAME ?= WROOM-02
And #make menu-config create sdkconfig correctly
but with #make y have this error:
App "esp-at" version: ESP-AT
LD build/esp-at.elf
/opt/esp/esp-at/components/at/lib/libesp8266_at_core.a(at_baseCmd.c.obj):(.literal.at_setupCmdSleep+0x0): undefined reference to `esp_pm_configure'
/opt/esp/esp-at/components/at/lib/libesp8266_at_core.a(at_baseCmd.c.obj): In function `at_setupCmdSleep':
(.text.at_setupCmdSleep+0x86): undefined reference to `esp_pm_configure'
/opt/esp/esp-at/components/at/lib/libesp8266_at_core.a(at_baseCmd.c.obj): In function `at_setupCmdSleep':
(.text.at_setupCmdSleep+0x93): undefined reference to `esp_pm_configure'
collect2: error: ld returned 1 exit status
I cannot find function void esp_pm_configure(*config) on ESP8266_RTOS_SDK
Any suggestion ? 
Thank you
Pablo Z.

xcguang, 2020-04-16:
It looks the version of ESP8266_RTOS_SDK is different with  `module_config/module_esp8266_default/IDF_VERSION`

pablinza, 2020-04-16:
I did not realize the mismatch version. 
 git cloning of ESP8266_RTOS_SK with --branch release/v3.3 solved the problem.  
thank you xcguang. :) :)

closed 2020-04-16 by pablinza

xcguang, 2020-04-17:
You'd better not set IDF_PATH, and if there is no esp-idf in the project,  it will automatically clone the correct SDK when compiling.