espressif/esp-at #397: custom bin for ESP8266_1M not work

pablinza, 2020-04-17:
Hello. 
I'm trying to build a AT firmare for ESP01 1Mb.
After to do:
- change Makefile to ESP8266_1MB
- set menuconfig->Component config->ESP8266-specific->UART for console->Default: UART0
- change pins TX=1 RX=3 in factory_param_data.csv an factory_param_type.csv
- #make
- #make flash
In serial terminal at 115200 show this:
boot: ESP-IDF v3.3-rc1-64-gfaae0a99-dirty 2nd stage bootloader [0m<\r><\n>
 [0;32mI (58) boot: compile time 11:03:28 [0m<\r><\n>
 [0;32mI (67) qio_mode: Enabling default flash chip QIO [0m<\r><\n>
 [0;32mI (67)  boot: SPI Speed      : 40MHz [0m<\r><\n>
 [0;32mI (68)  boot: SPI Mode       : QIO [0m<\r><\n>
 [0;32mI (72)  boot: SPI Flash Size : 1MB [0m<\r><\n>
 [0;32mI (76)  boot: Partition Table: [0m<\r><\n>
 [0;32mI (80)  boot: ## Label            Usage          Type ST Offset   Length [0m<\r><\n>
 [0;32mI (87)  boot:  0 otadata          OTA data         01 00 00009000 00002000 [0m<\r><\n>
 [0;32mI (94)  boot:  1 phy_init         RF data          01 01 0000f000 00001000 [0m<\r><\n>
 [0;32mI (102) boot: 2 nvs              WiFi data        01 02 00010000 00008000 [0m<\r><\n>
 [0;32mI (109) boot: 3 at_customize     unknown          40 00 00018000 00008000 [0m<\r><\n>
 [0;32mI (117) boot: 4 factory          factory app      00 00 00020000 000e0000 [0m<\r><\n>
 [0;32mI (124) boot: End of partition table [0m<\r><\n>
 [0;32mI (129) boot: Defaulting to factory image [0m<\r><\n>
 [0;32mI (133) esp_image: segment 0: paddr=0x00020010 vaddr=0x40220010 size=0xb2694 (730772) map [0m<\r><\n>
 [0;32mI (388) esp_image: segment 1: paddr=0x000d26ac vaddr=0x402d26a4 size=0x1ebb8 (125880) map [0m<\r><\n>
 [0;32mI (431) esp_image: segment 2: paddr=0x000f126c vaddr=0x3ffe8000 size=0x00694 (  1684) load [0m<\r><\n>
 [0;32mI (432) esp_image: segment 3: paddr=0x000f1908 vaddr=0x40100000 size=0x00a50 (  2640) load [0m<\r><\n>
 [0;32mI (439) esp_image: segment 4: paddr=0x000f2360 vaddr=0x40100a50 size=0x059cc ( 22988) load [0m<\r><\n>
 [0;32mI (455) boot: Loaded app from partition at offset 0x20000 [0m<\r><\n>
phy_version: 1156.0, 635e6fc, Feb 20 2020, 16:20:08, RTOS new<\r><\n>

But i dont have any response to sended"AT\r\n".
Im not sure if #make complete all steps.
"How_To_Set_AT_Port_Pin.md" step (3) says Recompile the `esp-at` project, download the new `factory_param.bin` and AT bin into flash. 
Where is this bin flie to download?

Thanks.

pablinza, 2020-04-17:
Now i can generate /build/factory/factory_ESP8266_1MB.bin with #make factory_bin
0x9000,ota_data_initial.bin
0x0,bootloader/bootloader.bin
0x18000,at_customize.bin
0x19000,customized_partitions/factory_param.bin
0x1a000,customized_partitions/client_cert.bin
0x1b000,customized_partitions/client_key.bin
0x1c000,customized_partitions/client_ca.bin
0x1d000,customized_partitions/mqtt_cert.bin
0x1e000,customized_partitions/mqtt_key.bin
0x1f000,customized_partitions/mqtt_ca.bin
0x20000,esp-at.bin
0x8000,partitions_at.bin

i download with #esptool --chip esp8266 write_flash --flash_mode dio 0x0000 factory_ESP8266_1MB.bin.

The terminal output show this:
[0;32mI (455) boot: Loaded app from partition at offset 0x20000[0m
phy_version: 1156.0, 635e6fc, Feb 20 2020, 16:20:08, RTOS new
2.0.0
max tx power=78,ret=0
ready
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.Core 0 register dump:
PC      : 0x402261df  PS      : 0x00000030  A0      : 0x402261dc  A1      : 0x3fff4120
A2      : 0x00000000  A3      : 0x402d50ac  A4      : 0x00000002  A5      : 0x00000000
A6      : 0x3fff0618  A7      : 0x00000002  A8      : 0x000000ff  A9      : 0x000000ff
A10     : 0x00000001  A11     : 0xffffffff  A12     : 0x3fff50a0  A13     : 0x402d50a8 A14     : 0x00000040  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c

:( :( Ohh.. ups

pablinza, 2020-04-17:
trying again with 
#rm sdkconfig
#make menuconfig to change port setting and flowcontrol ->0
#.make
#make flush 
is working.... :):) ... 
ready
AT+GMR
AT version:2.1.0.0-dev(518c1ce - Apr 13 2020 03:48:58)<\r>
SDK version:v3.3-rc1-64-gfaae0a99-dirty<\r>
compile time(128945f):Apr 17 2020 17:49:44<\r>
Bin version:2.0.0(Unknown)

thanks..........

closed 2020-04-17 by pablinza

edbek, 2020-11-14:
Strange, but I do not even have the "factory_param.bin" file. How is the file "factory_param.bin" generated?

pablinza, 2020-11-16:
You generate it using the following commands:
#make clean
#make
#make print_flash_cmd | tail -n 1 > build/download.config 
#make factory_bin
#make erase_flash
the file bin is the folder
{esp-at dir}/build/factory/

xcguang, 2020-11-17:
please use `./build.py` on the latest master branch

edbek, 2020-11-20:
Thank you, pablinza and xcguang. But for some reason, when I start ./build.py, it freezes and nothing is done.

xcguang, 2020-11-20:
Remove `build` directory, then  `./build.py  build`