espressif/esp-at #398: Using fatfs I can not mount the file system

agatrost, 2020-04-23:
I used the at_http_server example to set up the code to be able to load the fatfs partition in the
at_customize.  
The order I am of the functions I am using that return successful are:

esp_at_custom_partition_find(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_FAT, "/fatfs");

wl_mount(const esp_partition_t *partition, wl_handle_t *out_handle);

ff_diskio_register_wl_partition(BYTE pdrv, wl_handle_t flash_handle);

diskDrive[0] = (char)('0' + pdrv);
diskDrive[1] = ':';
diskDrive[2] = 0;

esp_vfs_fat_register(const char* base_path, const char* fat_drive, size_t max_files, FATFS** out_fs)

when I call :
f_mount (FATFS* fs, diskDrive, 1);

f_mount returns an error 13
/* (13) There is no valid FAT volume */

Are there any steps that I am missing in code?
Is there anything I have to do when making?
Any help or documentation would be helpful.

Let me know if there is anything else i need to supply to help get my issue resolved.

Thank you,
Andrew

ustccw, 2020-06-13:
@ESP-Coco could you please take a look

ESP-Coco, 2020-06-15:
Hi @agatrost Which version are you using? If v4.0, use the old SPI interface in the SPI configuration( Component config -> SPI Flash driver -> Use the lagacy implementation before IDF v4,0)

closed 2021-04-14 by FayeY