espressif/esp-at #399: busy p...

ravdeepboparai, 2020-04-25:
If you separate AT commands are sent simultaneously, it will reply "busy p...".
 
For example, I opened one socket and while it is processing it, I send request to open another socket
at 20:07:22.410 -> AT+CIPSTART=3,"TCP","192.168.2.18",5002

Log:
20:07:18.802 -> AT+CIPSTART=2,"TCP","192.168.2.11",5000
20:07:22.435 -> 
20:07:22.435 -> busy p...
20:07:27.898 -> 
20:07:27.898 -> ERROR

This is a major issue when we have to wait for Ping and we need to send data on socket.

ustccw, 2020-04-26:
@ravdeepboparai `esp-at` commands are serial

ravdeepboparai, 2020-04-26:
Yes, all AT engines are serial but they have Async and sync pair of commands. 
The commands starting with "+" are considered as Async replies, while OK and ERROR commands are considered as Sync.
For example, if I send a ping command with a timeout of 10 seconds, and I will be blocked to send data on an already opened socket for 10 seconds. 
Do I am missing some configuration in which I will be allowed to send data while ping is waiting for a reply?

ravdeepboparai, 2020-04-26:
This is something similar to https://github.com/espressif/esp-idf/issues/1271#issuecomment-352313549
In which a Host queue was implemented but still for a particular case, here in AT engine the response to the commands which goes over the internet should not block the commands to the device itself. This affects the purpose of the multicore, multithreaded environment of ESP32.

ustccw, 2020-04-29:
To be honest, i am sorry that i do not know what you need

closed 2020-08-18 by xcguang