espressif/esp-at #400: ESP32 setup Local TCP/SSL server.

hiflord, 2020-04-29:
Hello. I am trying to setup TCP server via SSL using with AT Commands. I did it without SSL. But now I want to setup that with SSL. How can I do that? Do you have any document about it?

ustccw, 2020-05-06:
@hiflord  please take a look, [docs](https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md)

hiflord, 2020-05-06:
I looked this doc but I need to an example about how to setup local ssl tcp server. Also I need to know how to check this crypted datas.

ustccw, 2020-05-06:
@hiflord I am no sure you read the wrong docs, please view the [ESP_AT_Commands_Set.md](https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md)  again,

there are some examples under the `4.10 AT+CIPSERVER`

ustccw, 2020-05-06:
you can capture the TLS packets to check the crypted datas, by wireshark, omnipeek, etc...

closed 2021-04-14 by FayeY