espressif/esp-at #402: How to save the SNTP time

tohyf, 2020-04-30:
I have an application whereby the ESP32 will be powered down after brief use. I use SSL so I use the CIPSNTPCFG command to set up the time on the ESP32 so that the certificates are valid. However, there is a long delay (from a few seconds or up to 10 secs depending on network condition) before I can start using SSL. How to save the SNTP time so that it doesn't always start from 1970 after booting up?

ustccw, 2020-05-07:
@tohyf Hi tohyf, it is not recommended to save SNTP time into flash, because it will be a risk when  the certificate reaches an expiry date nearby.

tohyf, 2020-05-07:
But if i don't save the SNTP time, the ESP32 will always start with January 1970 when power-up. It need 5-10seconds to update the time, before I can use SSL commands, which is too slow. It will power up at least several times per week, so it is ok even if certificate expire. We will update the CA certificate every several years.
I just don't want it to always go back to 1970, making all certs invalid.

ustccw, 2020-05-08:
it dose cost a long time. you can change or add the `SNTP server` by `AT+CIPSNTPCFG`.

tohyf, 2020-05-08:
Yes I am already using AT+CIPSNTPCFG. I tried several servers, even google NTP server. Sometimes it takes 1 second, sometimes 10 second to update the time, so it is not consistent. I want to avoid user keep waiting when the time sync in progress. My plan is to "defer" the time update or let it run in the background.

I just want the ESP32 to start with the time of the last time sync. Or can I send the date/time to ESP32? So that it is not 1970 when power up.

Current flow (too slow):
boot -> AT+CIPSNTPCFG -> use AT+CIPSNTPTIME?  to check time -> AT+CIPSTART(SSL) .... -> power off
What i want:
boot -> [program the ESP32 time to last known datetime] -> AT+CIPSTART (SSL) -> AT+CIPSNTPCFG -> AT+CIPSNTPTIME? -> Power off

I cannot control the power, it will get powered off after each use.

ustccw, 2020-05-09:
@tohyf Maybe here is a workaround for you, though there may be a potential risk for real system time.

Add the following code to `main/app_main.c`, call it at the end of app_main(), system will get a virtual time by settings.

```

#include <time.h>
int esp_boot_set_virtual_time(void)  
{
    struct tm _tm;
    struct timeval tv;
    time_t unix_timestamp;
    _tm.tm_sec = 14;     // second: 14
    _tm.tm_min = 13;     // minute: 13
    _tm.tm_hour = 12;    // hour: 12
    _tm.tm_mday = 9;    // day: 9
    _tm.tm_mon = 4;     // May
    _tm.tm_year = 120;  // year: 2020
  
    unix_timestamp = mktime(&_tm);
    tv.tv_sec = unix_timestamp;
    tv.tv_usec = 0;

    if (settimeofday(&tv, (struct timezone *)0) < 0) {
        printf("settimeofday error\r\n");
        return -1;
    }

    return 0;
}
```

ustccw, 2020-05-09:
[image]

xcguang, 2020-05-09:
You can try to supply power for the VDD3P3_RTC separately, and keep it on all the time.

tohyf, 2020-05-10:
@ustccw - Thanks for your suggestion. I can try that as a workaround, however, in the future, the date will become too old too... Unless i can save it to flash periodically?
@xcguang  - My application do not have control to the supply. It will be cut off when it is no longer needed, so I cannot keep it on all the time.

ustccw, 2020-05-11:
@tohyf yeah, you can save real time to flash by `nvs_xxx` APIs.

closed 2020-08-18 by xcguang