espressif/esp-at #404: esp32-wroom-32 ble communication

elettronica67, 2020-05-03:
Hi, I need to communicate raw rx-tx data between mobile phone and esp32 initialized using AT commands, don't need what profile/service will be used, I just need to send and receive raw data.
Unfortunately I need to use BLE because with the Classic bluetoot serial port profile I'm not able to communicate with apple phones.

Can someone give me an example?
Thanks in advance

xcguang, 2020-05-13:
`AT+BLESPP`?

elettronica67, 2020-05-15:
Thank you so much, I have studied this command but what profile have to set on the mobile phone or on the software? I know that BLE 4.x doesn't admit the Serial Port Profile, so I don't know how to set my software side....
If you have some example written in any language it could be really useful for me

xiewenxiang, 2020-06-04:
https://github.com/espressif/esp-at/blob/master/docs/ESP32_AT_Classic_Bluetooth.md

closed 2020-08-10 by xiewenxiang