espressif/esp-at #406: AT+BLEGATTSSRVCRE doesn't work

tohyf, 2020-05-07:
I want to use Bluetooth BLE in the ESP32. I am seeing this error even after I try to add the service bin:
[14:38:34:103] AT+BLEINIT=2
[14:38:34:724] 
[14:38:34:724] OK
[14:38:36:500] AT+BLEGATTSSRVCRE
[14:38:36:502] 
[14:38:36:504] ERROR

Method 1:
I added the required csv file to components/customized_partitions/raw_data/ble_data/, 
then in AT_partition_dependency, this line is set: "AT_BLE_COMMAND_SUPPORT=y ble_data"
then, I run "make flash" to build and flash the ESP32.
And I still get the same error.

Method 2:
I use this command to generate the bin :
python BLEService.py -t GattServiceExample.bin GattServiceExample.csv
Then I use esptool to flash: esptool.py write_flash 0x21000 GattServiceExample.bin
and still get the same error when running the AT+BLEINIT=2 and AT+BLEGATTSSRVCRE

What is the correct method? Did I miss any step?

tohyf, 2020-05-12:
Anyone? how to debug this issue?

tohyf, 2020-05-18:
Where is the c source codes of esp_at_ble_cmd_regist()? I didn't find any of BLE source code in this repo.

xiewenxiang, 2020-06-04:
Did you change the `ble_data` file? If you change this file, can you share it? Maybe the format is not right.

the source code is in the `lib`

tohyf, 2020-06-06:
This is the csv file that i use in the ble_data:
[GattServiceExample.zip](https://github.com/espressif/esp-at/files/4739340/GattServiceExample.zip)

xiewenxiang, 2020-06-08:
@tohyf 

I test this csv file. But it seems works fine.

/home/espressif/Desktop/p1.png

tohyf, 2020-06-09:
how to debug it? any debug logs that i can turn on?

sdrabb, 2020-07-19:
got same error...

xcguang, 2020-07-19:
Recently, we find the binary generated by python3.x is error, and the bug has not been fixed, could you provide the ble_data.bin?

sdrabb, 2020-07-19:
This is generated following https://github.com/espressif/esp-at/tree/master/tools and the .csv file is untouched.
[ble_data_old.zip](https://github.com/espressif/esp-at/files/4943521/ble_data_old.zip)

xcguang, 2020-07-19:
Could you generate the binary using python2.x? for example, `python2 SCRIPT_PATH -t TARGET_FILE_NAME SOURCE_FILE_NAME`

xcguang, 2020-08-24:
Now, python3 should works well on the latest release/v2.1.0.0_esp32 and master branch.

closed 2020-08-24 by xcguang