espressif/esp-at #407: ESP32  AT指令使用ble 有些手机很难配对

woshi520laoxie, 2020-05-08:
使用ESP32的ble蓝牙
1、有些手机很难连接。
2、某些时候手机扫描到蓝牙，但是也很难连接，有种不稳定的感觉。
期望只要时扫描到蓝牙，就能连接上，至少是两次就能连接上。
连接上的时候有些手机

如果连接上的时候，有些手机接收到
0> BLE_LINK -> +BLECONN:0,"4e:41:46:f5:2d:f0"
 0> BLE_LINK -> +BLECONNPARAM:0,0,0,6,0,500
 0> BLE_LINK -> +BLECONNPARAM:0,0,0,36,0,500

有些手机
 0> BLE_LINK -> +BLECONNPARAM:0,0,0,39,0,500

更新当前连接间隔为39 、36
不过和这个关系不大，因为好连接的时候都是好连接的，有时候蓝牙就是连接不上。

如何能让蓝牙易于连接呢？
因为这个严重影响到用户的体验了。
情形1：有些手机同时都扫描到了同一个蓝牙，比如蓝牙A，但是某一台手机连接不上，另外一台手机却很容易连接。
情形2：今天使用手机A连接蓝牙A，很容易连接，过两天，使用手机A连接蓝牙A却经常超时，有没有什么办法是可以提高蓝牙配对的成功率的。
或者需要提供哪些信息可以给你们得到有效的数据呢？
因为蓝牙的连接是被动的。目前只能得到蓝牙配对成功，或者没有反应。

xiewenxiang, 2020-05-08:
在蓝牙连接时候，wifi 有没有在工作？

woshi520laoxie, 2020-05-08:
正常使用的时候wifi是在工作的。不过现在我也有测试仅蓝牙在工作的时候，有些手机也很难配对
不是配对不上，是搜索到了，但是很难配对

woshi520laoxie, 2020-05-09:
@xiewenxiang 你好啊，我使用蓝牙抓包工具抓了一些数据包，有两个手机进行连接，现象都是不一样的。我看起小米手机由于某些情况无法建立连接，vivo手机，请求也有回应但是最终还是失败了。
[image]

xiewenxiang, 2020-05-11:
@woshi520laoxie 
可以提供下这几款手机的具体型号吗

woshi520laoxie, 2020-05-11:
@xiewenxiang 你好啊，这个其实这个和手机可能要影响也不是很大，因为同一台手机也有经常连接不上的情况，有时候需要连接几十次才能连接上，有时候每次都能连接上，但是连接不上的时候经常就很难连接，我看AT指令能配置的并不多，还不清楚具体原因是哪个导致的。

woshi520laoxie, 2020-05-11:
@xiewenxiang 之前我测试的时候，因为我使用vivox21连接很容易，但是小米5很难连接，但是过一段时间之后，vivox21也连接不上，华为也连接不上，但是小米5却连接上了。

woshi520laoxie, 2020-06-03:
@xiewenxiang 其实和手机没有多大的联系的，我发现某些时候手机的确连接不上ble，我使用开发板测试的时候是这样的
1、设置蓝牙ble为服务器
2、配置广播参数
3、开始广播
-----------  这种最小的数据量的情况下，蓝牙是可以连接的。
那么在esp32 ble能搜索到蓝牙，但是却无法连接的情况下， 有那种因数对这种有影响呢？

目前这问题，导致项目很难往下进行了。

xiewenxiang, 2020-06-04:
广播参数是多少？你那里有抓包卡吗？比如 TI 的packet sniffer，看看是否广播已经开启了，如果抓包卡能抓到 ESP32 发出的广播包，手机还是连不上，可能就是手机那里的问题

另外，你提到这个问题，只仅在你提到的这几款手机上出现吗？ 还是你只用了这几款手机，这几款手机都不行？

woshi520laoxie, 2020-06-04:
广播参数是多少？你那里有抓包卡吗？比如 TI 的packet sniffer，看看是否广播已经开启了，如果抓包卡能抓到 ESP32 发出的广播包，手机还是连不上，可能就是手机那里的问题
 
另外，你提到这个问题，只仅在你提到的这几款手机上出现吗？ 还是你只用了这几款手机，这几款手机都不行？
 
1、广播参数我不太清楚

2、我手上有个抓包工具，可以抓到广播包，连接的过程也看得到，的确是失败了。

3、目前苹果手机不行，有一位同事的华为手机也不行<只有一台不行，那么先不管一台手机是有问题&gt;

4、最大的问题是:有很多台手机都能连接上，但是有时候失败率非常高，连接很多次都不行，除非复位

&nbsp; &nbsp; &nbsp;我先不管那些一直连接不上的手机，下面的这些是可以连接的手机，问题在于，如果连接不上，那么就经常连接不上

xiewenxiang, 2020-06-04:
@woshi520laoxie 
可以提供下你的抓包log 吗

woshi520laoxie, 2020-06-04:
目前并没有专门的log包，你需要大概什么样子的log包，我可以抓一些出来，目前正是运行的code是没有任何调试信息的，需要调试信息，只能另外抓。你希望看到哪种类型的log包，包含哪些信息，我这边可以修改一个调试工程记录log。

xiewenxiang, 2020-06-04:
我想要的是抓包卡抓的包，你那里是有抓包卡的吧？

woshi520laoxie, 2020-06-04:
有的，可以在ubuntu上面进行抓包，之前抓过一些数据包，然后放到另外一台电脑上面了。现在也可以试着抓一些包下来，需要的话，我现在就抓一些数据包。

xiewenxiang, 2020-06-04:
好的。你把完整的连接失败的log，提供一份给我吧

woshi520laoxie, 2020-06-04:
抓包的数据，估计没那么容易抓，因为目前连接状态很好，只能等状态不好的时候抓到数据了，我截取下来

xiewenxiang, 2020-06-04:
好的，如果抓取到，在这里贴一下

woshi520laoxie, 2020-06-08:
@xiewenxiang 目前测试到华为的mate 10无法连接 蓝牙ble。一位同事的这台手机连接不上。能搜索到ble。

xiewenxiang, 2020-08-10:
@woshi520laoxie 

能扫描到，但是连接不上，也有一种可能性是，两个设备有 bonding 过，但是其中一段的 bonding 信息被删除了，另一端没有删除。 也可以检查下这一点，如果还新的信息同步，可以重新打开这个 issue.

closed 2020-08-10 by xiewenxiang