espressif/esp-at #408: Using ESP32, TCP CONNECT and IPD is very slow to return

tohyf, 2020-05-10:
Hi,
I use ESP-WROOM-32D to connect to a HTTP API (non-SSL) using the AT+CIPSTART and AT+CIPSEND.
However, I found that ESP32 takes 1s or even more for the whole process, from AT+CIPSTART to the +IPD data from server. From CIPSTART to "CONNECT OK" , it takes 500ms to 1000ms, and from CIPSEND -> SEND OK to the +IPD, it take 500ms or more, sometimes a few seconds.
When I test the same API on a laptop, connected to the same wifi, it only take about 50ms-100ms from request to response with server data.
Why is ESP32 so slow? How to debug the issue? Will it be faster if I directly write using the ESP IDF ? Is it the performance issue with ESP32?

ustccw, 2020-05-11:
could you please try to capture the packets, see what happened

tohyf, 2020-05-11:
@ustccw What packets? You mean the packets at the wifi router?
The ESP AT commands are normal, no errors.

ustccw, 2020-05-18:
@tohyf yeah, you can capture the 802.11 packets by `omnipeek` in windows, `wireshark` in mac or linux, etc..

From the packets, usually, it can be analyzed to find out which step is more time-consuming.

such at `AT+CIPSTART="TCP","www.baidu.com",80`, you can see the `SYN`, `SYN+ACK`, `ACK` packets timestamp in your packets.
[image]

In my captures, it takes about 15ms to build the TCP link in network.

closed 2021-04-14 by FayeY