espressif/esp-at #409: ESP32 AT2.0 TCP连接失去响应，卡在Zerowindow

RomeliFeng, 2020-05-10:
**1.模组版本**
AT+GMR
AT version:2.0.0.0(e44648c - Nov  4 2019 02:18:57)
SDK version:v3.3
compile time(7cac7ad):Nov  4 2019 06:11:40
Bin version:2.0.0(WROOM-32)
OK
**2.问题现象**
ESP32工作在TCP Client模式下
当使用默认CIPRECVMODE=0时。电脑端服务器连续往ESP32发送大量数据8k bytes / 100ms。ESP32在一段时间后进入不响应状态（接收不到+IPD的数据；电脑端服务器断开连接不响应，CIPSTATUS为3 TCP正常连接）
使用Wireshark抓吧分析发现ESP32在往电脑发送Zerowindow数据包（电脑端TCP服务器软件关闭后还能看到对ESP32 Zerowindow的回应，很奇怪换过几个软件都是一个现象）
这时重启电脑后查询CIPSTATUS状态为x 并且结尾为ERROR
**3.恢复方式**
 1） 使用CIPCLOSE断开TCP连接（不可行，无法区分连接是否正常）
 2）使用CIPRECVMODE切换为模式1，并且使用CIPRECVDATA读取出缓冲区内的数据（可以通过CIPRECVLEN查询到缓冲区内数据量为5744）
**4.问题分析**
在URC上传数据的时候，可能由于电脑端发送数据量太大。ESP32无法及时把数据通过IPD发送出去。然后缓冲区满，ESP32也不尝试发送IPD清空缓冲区。导致Zerowindo的死循环

RomeliFeng, 2020-05-10:
其他：为什么espressif的官网上没有最新的AT指令集版本。

ustccw, 2020-05-11:
@RomeliFeng Could you please providing the packets for us to look into what happened.

and how can i reproduce this issue, could you please describe all the steps, AT logs and AT commands you send, and the TCP server in PC, etc..

RomeliFeng, 2020-05-11:
**1.Background**
ESP32 TCP Client IP 10.10.1.184
Computrer TCP Server IP 10.10.1.99 Port 9999
Server send 3200Bytes dummy data to ESP32  interval 100ms
ESP32 will reply all data has received to Server
**2.ESP32 TCP Client log**
```
[00000002.102] D UStack.UAT_Client       CheckResponse: unknown urc->ready

[00000002.103] D UStack.UAT_Client       AT_Base: AT
[00000002.105] D UStack.UAT_Client       AT_Base: OK
[00000002.108] D UStack.UAT_Client       AT_Setup: AT+UART_CUR=2000000,8,1,0,0
[00000002.112] D UComm.UESP32_AT         UART_CUR_Setup: OK
[00000002.113] D UStack.UAT_Client       AT_Base: AT
[00000002.114] D UStack.UAT_Client       AT_Base: OK
[00000002.114] D UStack.UAT_Client       AT_Base: ATE1
[00000002.115] D UStack.UAT_Client       ATE_Base: OK
[00000002.116] D UStack.UAT_Client       AT_Query: AT+CWAUTOCONN?
[00000002.117] D UStack.UAT_Client       CheckResponse: unknown rsq->+CWAUTOCONN:0

[00000002.118] D UComm.UESP32_AT         CWAUTOCONN_Query: OK
[00000002.119] D UStack.UAT_Client       AT_Exec: AT+CWQAP
[00000002.120] D UComm.UESP32_AT         CWQAP_Exec: OK
[00000002.121] D UStack.UAT_Client       AT_Setup: AT+CWHOSTNAME="TY-009B"
[00000002.122] D UComm.UESP32_AT         CWHOSTNAME_Setup: OK
[00000002.123] D UStack.UAT_Client       AT_Setup: AT+CWJAP="xxx","xxx"

[00000002.288] W UModule.UESP32_AT1      URC_WIFI: ap connected

[00000003.733] D UStack.UAT_Client       CheckResponse: unknown rsq->WIFI GOT IP

[00000003.735] D UComm.UESP32_AT         CWJAP_Setup: OK

[00000010.000] D UStack.UAT_Client       AT_Setup: AT+CIPSTART="TCP","10.10.1.99",9999,10
[00000010.013] D UStack.UAT_Client       CheckResponse: unknown rsq->CONNECT

[00000010.014] D UComm.UESP32_AT         CIPSTART_TCP_Single_Setup: OK
[00000016.323] D UModule.UESP32_AT1      URC_IPD: receive data length->2920
[00000016.324] D UModule.UESP32_AT1      URC_IPD: receive data length->280
[00000016.325] D UStack.UAT_Client       AT_Setup: AT+CIPSEND=2048
[00000016.326] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: start trans data
[00000016.338] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: waiting sending
[00000016.339] D UStack.UAT_Client       CheckResponse: unknown rsq->Recv 2048 bytes

[00000016.341] D UComm.UESP32_AT         CIPSEND_Setup: OK
[00000016.341] D UStack.UAT_Client       AT_Setup: AT+CIPSEND=1152
[00000016.342] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: start trans data
[00000016.349] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: waiting sending
[00000016.350] D UStack.UAT_Client       CheckResponse: unknown rsq->Recv 1152 bytes

[00000016.352] D UComm.UESP32_AT         CIPSEND_Setup: OK

[00000017.755] D UModule.UESP32_AT1      URC_IPD: receive data length->2920
[00000017.757] D UModule.UESP32_AT1      URC_IPD: receive data length->280
[00000017.758] D UStack.UAT_Client       AT_Setup: AT+CIPSEND=2048
[00000017.759] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: start trans data
[00000017.770] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: waiting sending
[00000017.771] D UStack.UAT_Client       CheckResponse: unknown rsq->Recv 2048 bytes

[00000017.773] D UComm.UESP32_AT         CIPSEND_Setup: OK
[00000017.774] D UStack.UAT_Client       AT_Setup: AT+CIPSEND=1152
[00000017.775] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: start trans data
[00000017.782] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: waiting sending
[00000017.783] D UStack.UAT_Client       CheckResponse: unknown rsq->Recv 1152 bytes

[00000017.785] D UComm.UESP32_AT         CIPSEND_Setup: OK

[00000026.038] D UModule.UESP32_AT1      URC_IPD: receive data length->2920
[00000026.040] D UModule.UESP32_AT1      URC_IPD: receive data length->280
[00000026.041] D UStack.UAT_Client       AT_Setup: AT+CIPSEND=1792
[... 9 lines cut ...]

[00000026.067] D UComm.UESP32_AT         CIPSEND_Setup: OK

[00000026.717] D UModule.UESP32_AT1      URC_IPD: receive data length->2920
[00000026.718] D UModule.UESP32_AT1      URC_IPD: receive data length->280
[00000026.720] D UStack.UAT_Client       AT_Setup: AT+CIPSEND=2048
[00000026.721] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: start trans data
[00000026.732] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: waiting sending
[00000026.734] D UStack.UAT_Client       CheckResponse: unknown rsq->Recv 2048 bytes

[00000026.737] D UComm.UESP32_AT         CIPSEND_Setup: OK
[00000026.738] D UStack.UAT_Client       AT_Setup: AT+CIPSEND=1152
[00000026.741] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: start trans data
[00000026.747] D UComm.UESP32_AT         CIPSEND_TCP_Single_Setup: waiting sending
[00000026.748] D UStack.UAT_Client       CheckResponse: unknown rsq->Recv 1152 bytes

[00000027.056] D UStack.UAT_Client       CheckResponse: unknown rsq->+IPD,2920:01234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890 [... cut ...] 7890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789

[00000027.357] D UComm.UESP32_AT         CIPSEND_Setup: OK

```
3.Picture
[image]
[wireshark.pcapng.txt](https://github.com/espressif/esp-at/files/4609488/wireshark.pcapng.txt)
[Log.txt](https://github.com/espressif/esp-at/files/4609490/Log.txt)

ustccw, 2020-05-11:
It looks like `ESP32-AT` has a high throughput on TX and RX in short interval from your packets.

Maybe you can try to enlarge the following configuration.

- `make menuconfig` -> `Component config` -> `LWIP` -> `TCP` ->  `Default send buffer size` 
- `make menuconfig` -> `Component config` -> `LWIP` -> `TCP` -> `Default receive window size`

RomeliFeng, 2020-05-11:
Thanks, it may be can reduce frequency.But 治标不治本. The problem is since ESP32 rx buf full. I can't receive any more data. It's not normal. 
And may be add new AT command to set rx tx buffer?

xcguang, 2020-05-13:
可以采用最新的编译试一下

RomeliFeng, 2020-05-13:
> 可以采用最新的编译试一下

谢谢，有空我试试，但是未发布版本做产品我担心有别的bug。

closed 2020-06-02 by RomeliFeng