espressif/esp-at #410: ESP32 AT2.0 CIPSEND指令问题

RomeliFeng, 2020-05-10:
1.模组版本
AT+GMR
AT version:2.0.0.0(e44648c - Nov 4 2019 02:18:57)
SDK version:v3.3
compile time(7cac7ad):Nov 4 2019 06:11:40
Bin version:2.0.0(WROOM-32)
OK
2.问题现象
ESP32工作在TCP Client模式下
当使用默认CIPRECVMODE=0时。
通过CIPSEND指令发送数据时（此时电脑端还在往ESP32发送数据）
AT+CIPSEND=xxx
OK
//在这里有机会收到IPD的URC数据包
>
Recv xxxx

这个发送IPD URC数据包的位置非常不合理。本条指令还没结束怎么可以在执行中返回URC数据包。

ustccw, 2020-05-11:
@RomeliFeng  `AT+CIPSEND` has already returned `OK`, how do you think the command does not finish. Confused, Could you please describe in detail the process what you want?

RomeliFeng, 2020-05-11:
Hi. 
Although “AT+CIPSEND” has already returned “OK”, but until I received the symbol ">" and enter the data I want to send, I could not send the other AT commands(I coun't stop the transfer session).
At gernel GSM modules, such as Quectel M26, the AT command  CMGS is this logic:

```
AT+CMGS="15021012496"
> This is a test from Quectel // Insert message body，<CTRL+Z>Send，<ESC>Exit
+CMGS: 247 
OK
```
.
The other question is this command "AT+CIPSEND" has three ended response: "SEND OK", "SEND FAIL" and "ERROR", its different betwen gernel AT command "ERROR"(maybe has error code before) and "OK". this will make some special with command finish detect.

ustccw, 2020-05-11:
@RomeliFeng yeah, AT command is serial, the next command can only be sent after the last command has ended.

About `GSM modules`, i am not sure if the `<CTRL+Z>` and `<ESC>` will work well for all the serial tools, such as secureCRT, xshell, cutecom, etc..  
I guess not. A `<CTRL+Z>` interrupt can be capture and transmit to AT by special ASCII and do send or exit then.

About `AT+CIPSEND` responses, look into the similar GSM module M10, it is same to us. 

[image]

> Screenshot is from https://www.quectel.com/UploadFile/Product/Quectel_M10_AT_Commands_Manual_V4.0.pdf

RomeliFeng, 2020-05-11:
About the AT+CIPSEND response  i got it. As for the OK returned by AT + CIPSEND after sending, and the URC message what insert the AT+CIPSEND, I still feel weird.
Thank for you reply.

RomeliFeng, 2020-05-11:
@ustccw 
我发现这个IPD的URC消息不守规矩呀。他怎么能出现在别人回显的数据之后 指令完成之前啊。这样没法区分URC和正常回复的数据包了。
[image]

ustccw, 2020-05-18:
@RomeliFeng It looks like not the CIPSEND issue, could you please new a issue, describe your issue in detail.
what is you expected result, what is the now workflow.

In additional, please add the `AT+GMR` and some necessary steps for us to reproduce it.

closed 2020-05-18 by RomeliFeng