espressif/esp-at #412: ESP32_AT指令工程间隔一段时间编译会自动更新

woshi520laoxie, 2020-05-12:
ESP32_AT指令工程间隔一段时间编译会自动更新
1、问：能否取消这个自动更新，就用目前使用的版本就好。

ustccw, 2020-05-18:
@woshi520laoxie could you please providing some logs.  
what does the auto-update means?

woshi520laoxie, 2020-05-22:
@ustccw 
就是每次我编译 esp32_at工程的时候，就会自动更新，我想取消这个更新，因为每次更新也比较久，而且还不容易受控。

woshi520laoxie, 2020-05-22:
@ustccw

woshi520laoxie, 2020-05-22:
@ustccw 下面这个就是编译工程的时候自动更新，间隔一段时间更新 
make的时候就会自动更新 
[image]
期望是取消这个更新。不需要

woshi520laoxie, 2020-05-22:
@ustccw 我期望是我想更新的时候再进行更新，我不期望自动进行更新，可以把这个自动更新关闭吗？
下面这个就是 esp_at的工程，在这下面进行make的时候，如果间隔半个月回来再进行编译，可能就会自动进行更新操作了
[image]

ustccw, 2020-05-25:
@woshi520laoxie Do not `git pull` can disable the auto-update.  
But it is recommended to auto-update.

woshi520laoxie, 2020-06-01:
@ustccw 好的，谢谢

closed 2020-06-01 by woshi520laoxie