espressif/esp-at #413: virtual machine ubuntu1804.ova root password

Honzik321, 2020-05-14:
Hi, 
what is the root password in virtual machine ubuntu1804.ova ? I have no permission to flash the device /dev/ttyUSB0, I must change permission, but its not possible without root password.

Thank you

ustccw, 2020-05-18:
@Honzik321 the password is in the desktop background.

ustccw, 2020-05-18:
you can try with `1234567890`.

```
sudo chmod 777 /dev/ttyUSB0
```

Honzik321, 2020-05-18:
Thank you it works! 

BTW, I have explored the desktop background, the password is there but I don't speak Chinese language:
[image]

ustccw, 2020-05-18:
@Honzik321 thanks for you observation. It means the following:
- ubuntu admin password: `1234567890`
- run `./mount_share_dir.sh` on shell, mount the share directory to /mnt/share (please create share directory on windows D disk firstly)
- here are two projects in `~/project`, for `ESP32` and `ESP8266` platform. user can update the SDK by `git pull`

closed 2020-08-24 by xcguang