espressif/esp-at #414: how to change to pico-d4 module ?

Honzik321, 2020-05-14:
hi,

what is the correct process to change from wroom-32 to pico-d4 module ? There are different pinout for rx, tx, cts, rts between these modules. The configuration od wroom=32 doesnt work on pico=d4 module.

Thank you,
John

ustccw, 2020-05-18:
@Honzik321 Hi, the `PICO-D4` pins are configurated at `components/customized_partitions/raw_data/factory_param/factory_param_data.csv`

if you want to compile PICO-D4 fireware, you can set the following configuration at the root Makefile.
```
export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP32
export ESP_AT_MODULE_NAME ?=PICO-D4
```

Honzik321, 2020-05-18:
Thank you, I have changed to PICO-D4 in Makefile.  But no "ready" message after RESET is read on any pin of my ESP32-PICO-KIT_V4.1 module.

There is some warning during make defconfig: "Makefile:29: There is no module_config/module_pico-d4,Using module_config/module_esp32_default"

Could be the problem read from BOOT messages on UART TX0 ? If yes, could you help ? I send it below:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6952
load:0x40078000,len:14368
ho 0 tail 12 room 4
load:0x40080400,len:4248
entry 0x400806e0
[0;32mI (74) boot: Chip Revision: 1[0m
[0;32mI (74) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (42) boot: ESP-IDF v4.0-174-gf9cb434ee 2nd stage bootloader[0m
[0;32mI (42) boot: compile time 16:05:33[0m
[0;32mI (42) boot: Enabling RNG early entropy source...[0m
[0;32mI (47) boot: SPI Speed      : 40MHz[0m
[0;32mI (51) boot: SPI Mode       : DIO[0m
[0;32mI (55) boot: SPI Flash Size : 4MB[0m
[0;32mI (59) boot: Partition Table:[0m
[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (70) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (78) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (85) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (93) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (100) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (108) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (115) boot: End of partition table[0m
[0;32mI (120) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (127) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x20924 (133412) map[0m
[0;32mI (183) esp_image: segment 1: paddr=0x0012094c vaddr=0x3ffbdb60 size=0x03788 ( 14216) load[0m
[0;32mI (189) esp_image: segment 2: paddr=0x001240dc vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (190) esp_image: segment 3: paddr=0x001244e4 vaddr=0x40080400 size=0x0bb2c ( 47916) load[0m
[0;32mI (219) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0xff2fc (1045244) map[0m
[0;32mI (590) esp_image: segment 5: paddr=0x0022f31c vaddr=0x4008bf2c size=0x10618 ( 67096) load[0m
[0;32mI (618) esp_image: segment 6: paddr=0x0023f93c vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (635) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (635) boot: Disabling RNG early entropy source...[0m
2.0.0 max tx power=78,ret=0

ustccw, 2020-05-18:
@Honzik321 please test with the following firmware, test the pinout configuration is ready or not.

[factory_PICO-D4.zip](https://github.com/espressif/esp-at/files/4644312/factory_PICO-D4.zip)

Honzik321, 2020-05-18:
Hello, it works on pins 19 (TX) and 22 (RX)! Thank you!

AT+GMR
AT version:2.1.0.0-dev(b8d0d4d - May 13 2020 09:42:58)
SDK version:v4.0-327-gfe67bed
compile time(bffff9e):May 15 2020 05:38:57
Bin version:2.0.0(PICO-D4)

Where could be a mistake during my build process ?

Honzik321, 2020-05-18:
What I'm doing wrong during SLEEP command ? I have already tested AT+SLEEP on my SOLO module without success. Now, with PICO-D4 module and your BIN firmware I have still the same result:
----------------- AT Commands -----------------
ready
WIFI CONNECTED
WIFI GOT IP
AT+SLEEPWKCFG=1,1

OK
AT+SLEEP=1

ţX†f€€ń
---------------------------------------------------

The UART communication always stops to work after AT+SLEEP=1 or AT+SLEEP=2 or AT+SLEEP=3, even if I previously set the wake-up source to UART or GPIO. I must reset the module after the SLEEP command :-(

Thank you,
John

ustccw, 2020-05-19:
> Hello, it works on pins 19 (TX) and 22 (RX)! Thank you!
>
> AT+GMR
> AT version:2.1.0.0-dev(b8d0d4d - May 13 2020 09:42:58)
> SDK version:v4.0-327-gfe67bed
> compile [...]

Please try the following steps:
```
git clean -xdf && git reset --hard
export ESP_AT_PROJECT_PLATFORM=PLATFORM_ESP32
export ESP_AT_MODULE_NAME=PICO-D4
export ESP_AT_PROJECT_PATH=$(pwd)
./esp-idf/tools/idf.py build
cp build/flash_project_args build/download.config
make factory_bin
```

Honzik321, 2020-05-19:
Great! It works! Now I will test the sleep modes again.

Honzik321, 2020-05-20:
Hi, 

AT+SLEEP=2 (light sleep mode) still doesn't work. Could you please test it?  

////////////////////////////////////////////////////////////////////////////////////////
My AT commands:

ready
AT+SLEEPWKCFG=2,4,1

OK

AT+CWJAP="XXXX","XXXXXX","XX:XX:XX:XX:XX:XX",1,0,9
WIFI CONNECTED
WIFI GOT IP

OK
#                     -> about 27mA average current here
AT+SLEEP=2

OK
#                      -> about 10mA average current here

AT+CIPSTATUS
#                      -> 1. NO RESPONSE from UART here
#                      -> 2. aplying HIGH level on PIN4
#                      -> 3. about 27mA average current

AT+CIPSTATUS
Ě÷`řń
#                      -> This is response from UART after aplying high level on PIN4 (wake-up)
////////////////////////////////////////////////////////////////////////////////////////

I have also tested different wake-up sources with same result.

Thank you very much to solve this problem. I trying to solve it last month still without success.

John

ustccw, 2020-08-20:
@Honzik321 One issue for one question, please create a new issue.

closed 2021-04-14 by FayeY