espressif/esp-at #415: ESP32 Thing not responding AT commands

uilter, 2020-05-18:
Hi, 

I buy an ESP32 Thing board from Sparkfun.
I'd like to test the ESP-AT in this board to add a external MCU to my tasks.

After download the new firmware, compiled from scratch, I not able to get the AT responses from ESP32. I compiled from scratch because this board has external 26MHz XTAL.

I made the following changes at menuconfig:
- Disable flow control;
- Set XTAL to auto.

It's my serial monitor log at UART0:
> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:6896
> ho 0 tail 12 room 4
> load:0x40078000,len:14356
> load:0x40080400,len:4248
> entry 0x400806e0
> I (59) boot: Chip Revision: 1
> I (60) boot_comm: chip revision: 1, min. bootloader chip revision: 0
> I (43) boot: ESP-IDF v4.0-327-gfe67bedee-dirty 2nd stage bootloader
> I (43) boot: compile time 08:05:56
> I (52) boot: Enabling RNG early entropy source...
> I (52) boot: SPI Speed      : 40MHz
> I (53) boot: SPI Mode       : DIO
> I (57) boot: SPI Flash Size : 4MB
> I (61) boot: Partition Table:
> I (64) boot: ## Label            Usage          Type ST Offset   Length
> I (72) boot:  0 phy_init         RF data          01 01 0000f000 00001000
> I (79) boot:  1 otadata          OTA data         01 00 00010000 00002000
> I (87) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
> I (94) boot:  3 at_customize     unknown          40 00 00020000 000e0000
> I (102) boot:  4 ota_0            OTA app          00 10 00100000 00180000
> I (109) boot:  5 ota_1            OTA app          00 11 00280000 00180000
> I (117) boot: End of partition table
> I (121) boot_comm: chip revision: 1, min. application chip revision: 0
> I (128) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x2805c (163932) map
> I (195) esp_image: segment 1: paddr=0x00128084 vaddr=0x3ffbdb60 size=0x03960 ( 14688) load
> I (201) esp_image: segment 2: paddr=0x0012b9ec vaddr=0x40080000 size=0x00400 (  1024) load
> I (203) esp_image: segment 3: paddr=0x0012bdf4 vaddr=0x40080400 size=0x0421c ( 16924) load
> I (218) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0x10000c (1048588) map
> I (592) esp_image: segment 5: paddr=0x0023002c vaddr=0x4008461c size=0x17ef0 ( 98032) load
> I (633) esp_image: segment 6: paddr=0x00247f24 vaddr=0x400c0000 size=0x00064 (   100) load
> I (650) boot: Loaded app from partition at offset 0x100000
> I (650) boot: Disabling RNG early entropy source...
> BLUFI BLE is not connected yet
> 2.0.0
> max tx power=78,ret=0

AT command interface I just got a "ready" message.
If I try a simple "AT", it not respond.

Using a smartphone, I was able to connect on WIFI as showing the AT serial interface:
> +STA_CONNECTED:"20:28:5d:35:fb:51"
> +DIST_STA_IP:"20:28:5d:35:fb:51","192.168.4.2"

What am I doing wrong?

Thank you!

jwestmoreland, 2020-05-19:
@uilter ,

You must use a terminal program with the AT command I/F that either sends a <CR> or <LF> or both - depending on the compile options

Realterm is a good program for this - also YAT - https://sourceforge.net/projects/y-a-terminal/files/latest/download

You are getting the ready and your smartphone is connecting so you just need to use the correct command termination sequence.  The Hayes command set (AT) demarks the line input parsing on the termination character.

HTH,
John W.

uilter, 2020-05-19:
Hi @jwestmoreland thank you for your reply.

I already use the Tera Term, but thanks for the suggestion.
You are right, big mistake from my part. 
By default I use just CR and forgot to try it CR+LF.

Thank you for it.

closed 2020-05-19 by uilter

jwestmoreland, 2020-05-19:
@uilter ,

NP - glad it's resolved and you are welcome!
John W.