espressif/esp-at #416: Feature Request:  Persistent Transparent Xmission UDP Server w/Multiple Connections

jwestmoreland, 2020-05-19:
Hello,

A suggestion/request:

Is it possible to set up (maybe in future releases) a transparent UDP Server that will support multiple clients?  Maybe a transparent multicast type of protocol configuration?

Thanks In Advance,
John W.

ustccw, 2020-05-25:
@xcguang PTAL

jwestmoreland, 2020-05-25:
Hello @xcguang ,

Not sure if I understand the reply - please take another look - as I don't
think there's a true multicast server (yet) in the code that would handle
not only transparent persistent transmission and/or multiple target ports
as in
a true multicast protocol configuration; maybe I just haven't seen that yet
or found that yet.  Pretty sure from what I've looked at that mDNS won't do
actual true multicast - maybe I'm wrong.

Thanks,
John W.

xcguang, 2020-05-25:
Hi,
You can create an UDP connection with a specified port; and the remote UDP can send data to it. If the remote address is  multicast type, it will send multicast packet; as https://esp32.com/viewtopic.php?f=2&t=14247&hilit=multicast

jwestmoreland, 2020-05-25:
@xcguang ,

Yes - the point of the question is to have this be a part of ESP-AT, and
also be persistent across boots, and also be transparent - that's the
actual point of the question/request.

Unless I'm not seeing something - this is not part of the current ESP-AT
capability; hence the question/request.

Persistence and transparency across boots are part of this question and
request - which are two important points.

I'm not just simply asking for UDP multicast service - if I was, this
question would be posted under the esp-idf forum.

Thanks,
John W.

jwestmoreland, 2020-05-25:
@xcguang ,

Which build of ESP-AT will support multicast with the AT+CIPSTART command?
I don't see that in any of the current documentation.

I have a custom target board I designed since connectBlue's modules went
obsolete - so I can't use the build in the post you pointed out above.

Thanks,
John W.

jwestmoreland, 2020-05-25:
Hello Again,

OK - see it here:

SHA-1: 759cf597be22f6b8ee81238161fd4d84fb853cb7

* feat: Update at core version 38662e04

- feat: Optimize socket process
- feat: Optimize cipstatus command process
- feat: AT+CIPSTART support multicast for multicast address from 224.0.0.0
to 239.255.255.255
- feat: Add command for BT COD
- feat: Add command to support light sleep

Thanks,
John W.

xcguang, 2020-05-25:
The latest master branch should work well, you can try it.

jwestmoreland, 2020-05-25:
AT+CIPMODE=1
AT+CIPMODE=1
<CR>
ERROR<CR>

jwestmoreland, 2020-05-25:
AT+CIPSTART="UDP","239.192.1.2",0,123,0  <== ,2 here didn't work
AT+CIPSTART="UDP","239.192.1.2",0,123,0
local port:123<CR><CR>
change type:0<CR><CR>
CONNECT<CR>
<CR>
OK<CR>
AT+CIPMODE=1
AT+CIPMODE=1
<CR>
OK<CR>

jwestmoreland, 2020-05-25:
Running wireshark and not seeing any multicast packets being transmitted
from the ESP32...

jwestmoreland, 2020-05-26:
@xcguang ,

I fixed a routing issue on my test system and the ESP32 is receiving
multicast packets but it doesn't appear to be sending them - do I have to
enable mDNS or do something else to get the server working on the ESP32?

Thanks,
John W.

jwestmoreland, 2020-06-02:
@xcguang ,

Do I need any special set up to be able to send multicast packets from the ESP32 using the 4.0 compatible ESP-AT build?

Thanks,
John

xcguang, 2020-06-02:
Please set remote port parameter

xcguang, 2020-06-02:
[image]

jwestmoreland, 2020-06-07:
@xcguang ,

I am not seeing the +IPD,6:123455 in my log - am I missing something?

Thanks,
John

xcguang, 2020-06-07:
> Do I need any special set up to be able to send multicast packets from the ESP32 using the 4.0 compatible ESP-AT build?

Failed to send data or not received data?

jwestmoreland, 2020-06-08:
@xcguang ,

I have it working now - in a test mode configuration.  As soon as I have
more than one station reporting it's receiving the multicast packets OK -
I'll make a report - hopefully by the end of the day today.

Thanks,
John

jwestmoreland, 2020-06-08:
@xcguang ,

Multicast is working - I currently have one station broadcasting the multicast packets and two stations receiving - also - the
AT+SAVETRANSLINK command is working.

I suppose you guys can add this to your next round of docs as it's extremely useful.

Thanks for the help!
John W.

closed 2020-06-10 by igrr

jwestmoreland, 2020-06-11:
Wanted to add this screen shot showing multicast packets being received OK...
Regards,
John W.
[image]

ustccw, 2020-06-13:
@jwestmoreland We really appreciate your seriousness!