espressif/esp-at #417: How can I load a binary like hello-world.bin into a flash partition using AT commands?

espressifelse, 2020-05-21:
I need some help. I would like to write a binary file like hello-world.bin to a partition like ota_0 over UART0. Can I do this using AT commands like AT+FS and AT+SYSFLASH? The examples are not clear on how to do this.

This is my setup.
Toolchain version: esp-2019r2
Compiler version: 8.2.0
AT version:2.1.0.0-dev(b8d0d4d - May 13 2020 09:42:58)
SDK version:v4.0-327-gfe67bedee
compile time(0d586ac):May 20 2020 17:00:32
Bin version:2.0.0(WROVER-32)

ustccw, 2020-05-25:
@espressifelse Please take a look: 
https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md#cmd-FS
and 
https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md#cmd-SYSFLASH

espressifelse, 2020-05-25:
Yes, this limitation is blocking me.
> Users should not change the "name" and "type" of the user partitions which we defined in the "at_customize.csv". But "offset" and "size" can be changed, if necessary.

closed 2020-05-25 by espressifelse

espressifelse, 2020-06-02:
I found SYSFLASH would not work on the primary partition table, so I wrote a command to do what I need.

AT+WRITE_PARTITION

Set Command:
AT+COPY_PARTITION=destination,offset,length
Response:
OK

Parameters:
destination
0:	ota_0
1: 	ota_1
2:	ota_2
offset
offset in bytes, decimal. Must be increments of 4096.
length
1-4096: data length in bytes. Must be increments of 4096.

Example: Write 4096 bytes to ota_1 at offset 0
AT+WRITE_PARTITION=1,0,4096
[send your binary file of size 4096 bytes here, with no carriage return or line feed]
OK